MicrosoftDocs/windows-driver-docs #3098: GFLAGS.exe - Monitoring Silent Process Exit

edwin-alvarez, 2022-04-09:
I'm trying to run the GFLAGs on WIN10 and it does not look like is working anymore,

I can see the HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\Image File Execution Options\ProcessName\GlobalFlag getting set. same as HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows NT\CurrentVersion\SilentProcessExit\ProcessName

if I close and open the gflags I can see gflags remembers the previous settings. but when I kill the app under monitor using task manager nothing is occurring. 

[image]
[image]

Monitor process: 
powershell -executionpolicy unrestricted -File "C:\temp\message.ps1"

code inside the message.ps1 is:

```
Add-Type -AssemblyName System.Windows.Forms
[System.Windows.Forms.MessageBox]::Show('SC Host process crashed.  SC needs to be re-started.','SC crashed','Ok','Error')
```

tedhudek, 2022-04-11:
Hi @edwin-alvarez, thanks for the feedback.  Can you please provide a link to the documentation page you're referencing?

edwin-alvarez, 2022-04-11:
looks like this capability is no longer working 

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/registry-entries-for-silent-process-exit

if you can redirect me to the correct ms team that can help me enable this capability would be awesome.

DOMARS, 2022-05-06:
Hello @edwin-alvarez, thanks for passing on what you are seeing with your use of GFlags.exe and silent exit. I was able to test this with notepad and I was able to trigger the notification on exit, so I believe that it continues to be functional.
I did not check the "Ignore Self Exits" box in my testing. And I used a Windows shipped EXE that seems to meet the criteria mentioned in https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/registry-entries-for-silent-process-exit. _The monitoring feature does not detect normal process termination that happens when the last thread of the process exits. The monitoring feature does not detect process termination that is initiated by kernel-mode code._
So perhaps your app is not meeting the criteria for the functionality that GFlags offers, for the type of exits that it monitors.

For the best response going forward, I would recommend one of the forums where other driver developers and Microsoft engineers can be found. For example, the [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html). You can post your question there, perhaps with a snip of the code that you are working on,  and the community of developers usually give feedback fairly quickly. Good luck with your use of gflags.

closed 2022-05-06 by DOMARS