MicrosoftDocs/windows-driver-docs #3099: No Driver Update Acceptable (DUA) process available for Certification Type (Attestation)

GuidoKiener, 2022-04-11:
Why is it not possible to update drivers signed with attestation signing? The https://partner.microsoft.com/de-de/dashboard/hardware is quite confusing when many driver updates are uploaded with "Submit new hardware".

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3ef5898d-310c-6966-70f9-929b96a3182e
* Version Independent ID: c4eb1177-38fc-d706-15ac-f48215c54c95
* Content: [Manage hardware submissions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/manage-your-hardware-submissions)
* Content Source: [windows-driver-docs-pr/dashboard/manage-your-hardware-submissions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/manage-your-hardware-submissions.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

anaharris-ms, 2022-04-11:
@GuidoKiener  

Thanks for notifying us of this issue. I am looking into this and will report back when I get more information.

anaharris-ms, 2022-04-18:
@GuidoKiener 
DUA is not available for attestation signed drivers.

anaharris-ms, 2022-04-19:
@GuidoKiener The reason is that DUA requires HLK. However, thank you for your feedback. I have communicated your concerns to the team.

closed 2022-04-19 by GuidoKiener