MicrosoftDocs/windows-driver-docs #3101: Confirm the steps to realize the registration of mobileoperatornotification

ZaleYan, 2022-04-13:
[Enter feedback here]
I want to develop the mobileoperatornotification for a C++ Service, But I can not found how to registry the mobileoperatornotification. Please help to confirm the steps to achieve registration

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a738277f-1f8a-a60e-a6ef-c33262d813cf
* Version Independent ID: fe96e047-6a43-62d9-7667-28cd821c1169
* Content: [Develop an app to handle the MobileOperatorNotification event - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/develop-an-app-to-handle-the-mobileoperatornotification-event)
* Content Source: [windows-driver-docs-pr/mobilebroadband/develop-an-app-to-handle-the-mobileoperatornotification-event.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/develop-an-app-to-handle-the-mobileoperatornotification-event.md)
* Product: **windows-hardware**
* Technology: **mobilebroadband**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2022-04-13:
Hi @ZaleYan, 
Thanks for your question. Please see [Register for the MobileOperatorNotification event by using metadata](https://docs.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/mobile-operator-notification-event-technical-details#register-for-the-mobileoperatornotification-event-by-using-metadata). If you still have questions, please open a new issue.

closed 2022-04-13 by aviviano

ZaleYan, 2022-04-14:
I have seen this method, but I use the C + + console app here. According to my understanding, I should not be able to register in the way you provide.

ZaleYan, 2022-04-14:
> Hi @ZaleYan, Thanks for your question. Please see [Register for the MobileOperatorNotification event by using [...]

I have seen this method, but I use the C + + console app here. According to my understanding, I should not be able to register in the way you provide.