MicrosoftDocs/windows-driver-docs #3105: ApiValidator.exe not found

YashMJ, 2022-04-22:
I have a visual studio project which is checking for ApiValidator.exe, which is not found in my system.
How can we install these dependent drivers in windows.

DOMARS, 2022-04-22:
@YashMJ - This particular channel is monitored by the men and women who write the driver documentation posted on docs.microsoft.com/drivers. Though we'd like to, we don’t have the resources to provide support for specific scenarios. It is difficult given the information provided to determine why Visual Studio, would require API Validator.  It is possible that this could be related to the APIValidator test that is included in the HLK - https://docs.microsoft.com/en-us/windows-hardware/test/hlk/testref/df4a9671-c2aa-4c81-b964-7247fb4799df  The HLK is a complex testing environment used to validate Windows drivers. 
Your best bet is to use one of the forums where other driver developers and Microsoft engineers can be found. For example the [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html). You can post your question there, and the community of developers usually give feedback fairly quickly.  Good luck with your use of the APIValidator in Visual Studio.

closed 2022-04-22 by DOMARS

y-m-j, 2022-04-22:
Thanks for the reply @DOMARS 
Solution to my own question : Installing Windows Driver Kit : https://go.microsoft.com/fwlink/?linkid=2166289
I just figured it out a bit late.