MicrosoftDocs/windows-driver-docs #3110: Connect Redirection change local IP address

zeltrax00, 2022-04-25:
I have tested and found out I can change local IP address in connect-redirect layer FWPM_LAYER_ALE_BIND_REDIRECT_V4 on Windows 10 (proof by TCPView). 

Can you confirm that this feature work on Windows 10 ? Because the document said it's not supported in connect-redirect layer https://docs.microsoft.com/en-us/windows-hardware/drivers/network/using-bind-or-connect-redirection

aviviano, 2022-04-28:
Hi @zeltrax00,
Can you please clarify your question? The documentation says that UDP packets **are** supported for redirection.
The layer at which redirection is performed determines the effect of the change. Changes at connect layers affect only the flow being connected. Changes at bind layers affect all connections that are using that socket.
The redirect layers are only available for Windows 7 and later versions of Windows.

zeltrax00, 2022-04-29:
Hi @aviviano , the documentation says `Changing the local address and port of a flow is only supported in the bind-redirect layer. This is not supported in the connect-redirect layer.`. But from my experiment, I can successfully change local address and port of a TCP connect in connect-redirect layer, but not UDP. So I want to know if the documentation is out of date or it's just a bug.

aviviano, 2022-05-02:
Hi @zeltrax00,
I spoke with the product team, and they confirmed that the documentation is correct. They are aware that changing the local address and port of a flow can be done in TCP, but it is still not supported.
Thanks for your question.

closed 2022-05-02 by aviviano

zeltrax00, 2022-05-20:
> 
I have few more question:
1. So what do I need to do if I want to change the interface for UDP packet after `bind` phase ?
2. Does the example code in the above documentation supports UDP or it is an TCP example.

zeltrax00, 2022-05-20:
> Hi @zeltrax00, I spoke with the product team, and they confirmed that the documentation is correct. They are aware that changing the local address and port of a flow can be done in TCP, but it is still not supported. Thanks for your question.

I have few more questions:
1. What am I supposed to do if I want to change the interface for UDP packet after `bind` phase ?
2. Does the example in the above documentation support UDP or it is a TCP example ?