MicrosoftDocs/windows-driver-docs #3124: TraceView no located at "tools\<Platform> subdirectory"

GF-Huang, 2022-05-05:
https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/traceview.md

[image]

[image]

And the WDK folder has no `tools` folder.

DOMARS, 2022-05-06:
@GF-Huang - Thanks for passing on what you are seeing with the lack of the "tools" directory. I looked on three PCs where I have the WDK installed and the TraceView utility, was located in the tools directory. Perhaps there is an issue with the installation of the WDK in your environment.  It is important to note that the installation and use of the WDK requires the installation of the SDK as well as Visual Studio. These pre-requisites are described in [Download the Windows Driver Kit (WDK)](https://docs.microsoft.com/windows-hardware/drivers/download-the-wdk). 

For further troubleshooting of your setup, your best bet is to use one of the forums where other driver developers and Microsoft engineers can be found. For example the [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html). You can post your question there, and the community of developers usually give feedback fairly quickly. Good luck with your use of the TraceView utility.

closed 2022-05-06 by DOMARS

GF-Huang, 2022-05-08:
I'm install WDK in a VM for testing, so I have not install VS in it.