MicrosoftDocs/windows-driver-docs #3144: Error on vs2022 startup after WDK installation

OsmanAltun, 2022-05-25:
This error occurs on startup of vs2022 version 17.2.2 after installing SDK and WDK (10.0.22621.1). Building sample also fails. 

```xml
  <entry>
    <record>784</record>
    <time>2022/05/25 14:02:41.934</time>
    <type>Error</type>
    <source>VisualStudio</source>
    <description>SetSite failed for package [Microsoft.Windows.Tools.WinIDE.Debugger.DebuggerPackage, DebuggerPackage, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null]Source: &apos;mscorlib&apos; Description: The object&apos;s type must be __ComObject or derived from __ComObject.&#x000D;&#x000A;Parameter name: o&#x000D;&#x000A;System.ArgumentException: The object&apos;s type must be __ComObject or derived from __ComObject.&#x000D;&#x000A;Parameter name: o&#x000D;&#x000A;   at System.Runtime.InteropServices.Marshal.ReleaseComObject(Object o)&#x000D;&#x000A;   at Microsoft.Windows.Tools.WinIDE.Debugger.DebuggerPackage.&lt;LoadDebuggerPackageOptions&gt;d__62.MoveNext()&#x000D;&#x000A;--- End of stack trace from previous location where exception was thrown ---&#x000D;&#x000A;   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()&#x000D;&#x000A;   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   at Microsoft.Windows.Tools.WinIDE.Debugger.DebuggerPackage.&lt;InitializeAsync&gt;d__52.MoveNext()&#x000D;&#x000A;--- End of stack trace from previous location where exception was thrown ---&#x000D;&#x000A;   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()&#x000D;&#x000A;   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   at Microsoft.VisualStudio.Shell.AsyncPackage.&lt;&gt;c__DisplayClass20_0.&lt;&lt;Microsoft-VisualStudio-Shell-Interop-IAsyncLoadablePackageInitialize-Initialize&gt;b__1&gt;d.MoveNext()&#x000D;&#x000A;--- End of stack trace from previous location where exception was thrown ---&#x000D;&#x000A;   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()&#x000D;&#x000A;   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   at Microsoft.VisualStudio.Threading.JoinableTask.&lt;JoinAsync&gt;d__76.MoveNext()&#x000D;&#x000A;--- End of stack trace from previous location where exception was thrown ---&#x000D;&#x000A;   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()&#x000D;&#x000A;   at Microsoft.VisualStudio.Services.VsTask.RethrowException(AggregateException e)&#x000D;&#x000A;   at Microsoft.VisualStudio.Services.VsTask.InternalGetResult(Boolean ignoreUIThreadCheck)</description>
    <guid>{22686B14-DA89-4978-A469-B4F8A565E2C9}</guid>
    <hr>80070057 - E_INVALIDARG</hr>
    <errorinfo></errorinfo>
  </entry>
```
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **whats-new-and-downloads**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

czp541308303, 2022-05-26:
I also encountered this problem and look forward to a solution

MiroKaku, 2022-05-27:
I also encountered this problem and look forward to a solution.

zodiacon, 2022-05-31:
Same issue here. Curiously enough, it doesn't happen on a VM I have VS installed on.
It seems to be related to the WDK plugin. Doing some analysis, the dialog that asks to press Yes/No to continue loading after an extension failed, but the window is INVISIBLE (I see it using my WinSpy tool, also Spy++). VS seems to be stuck on WaitForMultipleObjects where the only handle is the handle to the current thread.... very weird.

tedhudek, 2022-06-03:
Hi folks, apologies for the delay on this one.  We really appreciate the detailed feedback - I've conveyed it to the right people.  We have indeed had a regression -- I've added a note to the top of https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk with the details.  Keeping this issue open for now as we continue to work through this

smartmaster, 2022-06-03:
me too
I also encountered this problem and look forward to a solution.

chindakeodouangsy, 2022-06-03:
I encountered this problem as well, when install WDK on windows 11.

vikaspachisia, 2022-06-06:
Even I am facing this problem. I understand the issue exists and is also listed on WDK home page (https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk) but when will we get the resolution. Installing 60GB+ of VS is not a small task to revert to previous version of VS. That workaround is not simplistic and an extreme measure.

Cymoki, 2022-06-07:
Hey folks,
I’m a PM on the WDK team working to get a resolution for this issue, at this time an ETA for the resolution is not available. Once that becomes available, I’ll follow up with that information. I agree that rolling back is not simple and extreme, are there any issues with using WinDBG for a debugger solution?

carl-woodward, 2022-06-07:
Using WinDBG works fine for me; it is my normal flow (though I use both). Looking forward to the fix though.

Cymoki, 2022-06-08:
Hey Carl, you also mentioned that you're seeing a failure with ntdkk.h not found. Is that still reproducing?

vikaspachisia, 2022-06-08:
WinDBG works and I have been using both. But I am missing the remote debug experience provided by Remote Tools. I am a fan of integrated experience and VS. I have been using this heavily and missing it especially I get the feeling of reduced productivity. I understand this will take sometime but wanted to know this time as that can help me decide to continue to use windbg or roll back the VS version.

closed 2022-06-17 by tedhudek