MicrosoftDocs/windows-driver-docs #3150: missing class guid

Garthjag, 2022-05-31:
There seems to be a number of missing ClassGuid for example 
{1ed2bbf9-11f0-4084-b21f-ad83a8e6dcdc}
{4d36e966-e325-11ce-bfc1-08002be10318}
{533c5b84-ec70-11d2-9505-00c04f79deaf}
{62f9c741-b25a-46ce-b54c-9bccce08b6f2}
{c166523c-fe0c-4a94-a586-f1a80cfbbf3e}

When will these be updated on this page?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c231a8b4-24a8-c339-8660-52a2b9d28226
* Version Independent ID: 093b0d00-46ae-b685-760c-c04e10b27595
* Content: [System-Defined Device Setup Classes Available to Vendors - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/system-defined-device-setup-classes-available-to-vendors)
* Content Source: [windows-driver-docs-pr/install/system-defined-device-setup-classes-available-to-vendors.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/system-defined-device-setup-classes-available-to-vendors.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-06-02:
Hi @Garthjag, thanks for the feedback.

The second and third GUIDs that you list (4d... and 53...) are both listed on https://docs.microsoft.com/en-us/windows-hardware/drivers/install/system-defined-device-setup-classes-reserved-for-system-use, which is a page that forms a pair with the page you found, https://docs.microsoft.com/en-us/windows-hardware/drivers/install/system-defined-device-setup-classes-available-to-vendors, to list the Class and ClassGuid values.

I am looking into the other 3.

tedhudek, 2022-06-02:
It turns out the other 3 should all be on the "reserved for system use" page as well.  I will make that change sometime soon.

tedhudek, 2022-06-03:
Just pushed the doc update, should be on the live site soon.  Thanks again for the report!

closed 2022-06-03 by tedhudek