MicrosoftDocs/windows-driver-docs #3169: Replace Driver option greyed out

jbirdd1, 2022-06-18:
Hello,

I have submitted my Base Package to Microsoft according to the [Extension INF instructions](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/submit-dashboard-extension-inf-files#submitting-and-publishing-extension-infs). But I am having trouble figuring out how to create the Extension INF package. The “Replace Driver” option is greyed out for me. I’ve tried with an unsigned package, signed package, and the finalized submission packages. When I right click on the driver folder address under the Package tab, the option to "Replace Driver" is greyed out.

What am I missing?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: df0a9a8d-f361-42ac-e2b5-a261cc8bf866
* Version Independent ID: e32caa38-e43b-c607-9bdc-b07bba983e14
* Content: [Working with Extension INFs in the Windows Hardware Dashboard - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/submit-dashboard-extension-inf-files?source=docs#submitting-and-publishing-extension-infs)
* Content Source: [windows-driver-docs-pr/dashboard/submit-dashboard-extension-inf-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/submit-dashboard-extension-inf-files.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

anaharris-ms, 2022-06-20:
Hi @jbirdd1,

Navigate to [Services Hub](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fsupport.serviceshub.microsoft.com%2Fsupportforbusiness%2Fcreate&data=05%7C01%7Canaharris%40microsoft.com%7Cce0a1a4f4ec943812fdd08da52d82663%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637913385564313910%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=rX5usEQuGwfFgwsR1Io0tM1XSeZYA3ZrRUgZG8hjV0w%3D&reserved=0)

•	Select Contact us
•	Select the product family: Developer Tools
•	Developer Support topic: HLK\HCK
•	What product or service do you need help with: Windows Driver kit (WDK/HLK)
•	Select the product version: Hardware Lab Kit
•	Which category best describes the issue: Unable to create an Extension INF package.

anaharris-ms, 2022-08-22:
@Kevin-Tremblay do you know why this would be greyed out here?  Let me know if there is a scenario we are missing in the instructions.

Kevin-Tremblay, 2022-08-22:
+ @Cymon ***@***.***>, @Fidaa ***@***.***>

Replace Driver option greyed out * Issue #3169 * MicrosoftDocs/windows-driver-docs (github.com)<https://github.com/MicrosoftDocs/windows-driver-docs/issues/3169>

Sounds like HLK challenge. I'm guessing best way forward is to reply back with instructions to open a ticket?

jmmorto, 2022-08-22:
I found some MUCH better instructions on replacing the driver. https://docs.microsoft.com/en-us/windows-hardware/test/hlk/user/create-a-driver-only-update-package

Basically, you have to submit the base driver to Microsoft first. Then download the DUA Shell, launch HLK Studio by itself (not by clicking on a project), browse to and open your project, select "Open as Driver Update package", and then the Replace Driver option will show up.

Kevin-Tremblay, 2022-08-22:
Thanks Kevin!

We need a bit more information about the steps and what have been done to get to this point, the next step here is to file an incident with CSS team to get this info and work with the customer.

Here are the steps on how to create a case with CSS:

Windows developer support - Windows app development | Microsoft Developer<https://developer.microsoft.com/en-us/windows/support/?tabs=keytopics>

  *   Select Contact us
  *   Select Technical Support - Coding/Debugging
  *   Driver Development and Testing/Certification --> Submit an incident
  *   Fill in the required fields
  *   ***@***.***

Thanks
Fidaab

anaharris-ms, 2022-09-02:
Closing this, as the information seems to have been found. I confirmed that the link to this page is in the driver update page of the dashboard content: https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-update

closed 2022-09-02 by anaharris-ms