MicrosoftDocs/windows-driver-docs #3185: Incorrect Instructions

Sean-YHC, 2022-06-26:
The instructions do NOT work!!

There is no way to get to the page "Registration - Get a code signing certificate." When I pressed the NEXT in the previous page, it goes back to the previous page and never moving!!

After following someone's post to skip the page, and reaching the end (Review page), I still get an error message "Sorry, something went wrong . . ."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ed99e688-e27a-6b0f-0968-5f7d3117a73d
* Version Independent ID: ed99e688-e27a-6b0f-0968-5f7d3117a73d
* Content: [Register for the Microsoft Windows Hardware Developer Program - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-program-register.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-program-register.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

anaharris-ms, 2022-07-06:
@Sean-YHC Sorry you are having trouble with registration. Can you provide me with the exact steps you are taking, so we can find the problem you are having.

Sean-YHC, 2022-07-06:
@anaharris-ms I just followed the document (https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register). 

I have a EV code signing certificate. I also signed in with our organization's Azure AD Global administrator account.

Page 1: Get started with the Hardware Developer Program (No problem here)
Page 2: Registration - account details (no program to fill in the data, but the NEXT button does not work! It stays at the same page)

To move forward, someone recommended to go to: (https://partner.microsoft.com/en-us/dashboard/Registration/Hardware?step=SignUpload). It works. It jumps to Page 3.

Page 3: Registration - Sign and Upload Your File (No problem here)
Page 4: Registration - Review (No problem here. Account info and EV certificate from the page look good. Accept the terms, Press FINISH)
Page 5: Get the following message "**Sorry, something went wrong. Please try again later.**" 

The situation has not changed for the last two weeks. There are two places broken. The most important one is why there is something wrong??!!

anaharris-ms, 2022-07-06:
Her is what he wrote:

s<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fanaharris-ms&data=05%7C01%7Canaharris%40microsoft.com%7C22f5a1cfe02f4bdae78708da5f6f161a%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637927227964848204%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=ti1sbtipaoHBZMtONBJKOf9EIikd9M5TocAWecmnH%2B8%3D&reserved=0> I just followed the document (https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2Fen-us%2Fwindows-hardware%2Fdrivers%2Fdashboard%2Fhardware-program-register&data=05%7C01%7Canaharris%40microsoft.com%7C22f5a1cfe02f4bdae78708da5f6f161a%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637927227965004433%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=hjNiIbmsEAxVXkttjZGUjobXqiw1XnEbqNo1q3rvnp8%3D&reserved=0>).

I have a EV code signing certificate. I also signed in with our organization's Azure AD Global administrator account.

Page 1: Get started with the Hardware Developer Program (No problem here)
Page 2: Registration - account details (no program to fill in the data, but the NEXT button does not work! It stays at the same page)

To move forward, someone recommended to go to: (https://partner.microsoft.com/en-us/dashboard/Registration/Hardware?step=SignUpload<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fpartner.microsoft.com%2Fen-us%2Fdashboard%2FRegistration%2FHardware%3Fstep%3DSignUpload&data=05%7C01%7Canaharris%40microsoft.com%7C22f5a1cfe02f4bdae78708da5f6f161a%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637927227965004433%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=Xbf2Z8eS2zhHjRx1mSq4s%2BI15CGLdxOEfYlFuNh%2B84A%3D&reserved=0>). It works. It jumps to Page 3.

Page 3: Registration - Sign and Upload Your File (No problem here)
Page 4: Registration - Review (No problem here. Account info and EV certificate from the page look good. Accept the terms, Press FINISH)
Page 5: Get the following message "Sorry, something went wrong. Please try again later."

The situation has not changed for the last two weeks. There are two places broken. The most important one is why there is something wrong??!!

anaharris-ms, 2022-09-19:
@Sean-YHC ,

I've confirmed that this bug is not documentation related. If you are still having issues, please [contact technical support](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support#technical-support-issues).

Thank  you!

closed 2022-09-19 by anaharris-ms