MicrosoftDocs/windows-driver-docs #3194: New bugcheck code

LuigiBruno, 2022-07-09:
I've analyzed a Windows memory dump file by using WinDBG which returned the below information

Unknown bugcheck code (1aa)
Unknown bugcheck description
Arguments:
Arg1: 00000039d76edad0
Arg2: 0000000000000003
Arg3: ffffd680fd09e900
Arg4: fffffc03cd5b1598

I could not find any information about this bugcheck code.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ddc632b2-f4e1-b784-dd1e-878feac077f9
* Version Independent ID: 7a6866f4-3333-f2d1-86d7-dbb3d00925e0
* Content: [Bug Check Code Reference - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-code-reference2)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-code-reference2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-code-reference2.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2022-08-01:
@LuigiBruno - I'm working with the dev team to document this bug check code as well as 0x1ab which are related to incorrect stack memory access.

LuigiBruno, 2022-08-03:
> @LuigiBruno - I'm working with the dev team to document this bug check code as well as 0x1ab which are related to incorrect stack memory access.

OK @DOMARS.
Thank you.

DOMARS, 2022-08-23:
@LuigiBruno - This new bug check topic is now live and describes the bug check and its parameters. 

[Bug Check 0x1AA: EXCEPTION_ON_INVALID_STACK](https://docs.microsoft.com/windows-hardware/drivers/debugger/bug-check-0x1aa-exception-on-invalid-stack)

These parameter descriptions should also appear in a future release of !analyze.  Thanks for your interest in bug checks and helping making Windows more reliable.

closed 2022-08-23 by DOMARS