MicrosoftDocs/windows-driver-docs #3202: Missing Symbol

Jacksole, 2022-07-14:
Where can I find the symbols for nt! TRIAGE_9F_POWER. I get 
Symbol nt!TRIAGE_9F_POWER is not found when I run this program. Please help.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 28a05e45-339b-d94e-f337-e802afd56b79
* Version Independent ID: 8db6253e-eb0c-e88b-bf1e-7270a56096ca
* Content: [Bug Check 0x9F DRIVER_POWER_STATE_FAILURE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x9f--driver-power-state-failure)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x9f--driver-power-state-failure.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x9f--driver-power-state-failure.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

grahamm78, 2022-07-27:
This symbol should aready be included in the publicly available symbol for ntoskrnl  on https://msdl.microsoft.com/download/symbols
One thing to try, add an underscore to the beginning of the type name. many NT symbols use a typedef and you may need to use the original struct name which begins with an underscore.
dt nt!_TRIAGE_9F_POWER

0:000> lm
start             end                 module name
00000001`40000000 00000001`41046000   ntoskrnl   (pdb symbols)          C:\ProgramData\Dbg\sym\ntkrnlmp.pdb\5D6312DA6921E3A4E7F938B88330B0771\ntkrnlmp.pdb
0:000> dt TRIAGE_9F_POWER
Symbol TRIAGE_9F_POWER not found.
0:000> dt _TRIAGE_9F_POWER
ntoskrnl!_TRIAGE_9F_POWER
   +0x000 Signature        : Uint2B
   +0x002 Revision         : Uint2B
   +0x008 IrpList          : Ptr64 _LIST_ENTRY
   +0x010 ThreadList       : Ptr64 _LIST_ENTRY
   +0x018 DelayedWorkQueue : Ptr64 _TRIAGE_EX_WORK_QUEUE
   +0x020 DelayedIoWorkQueue : Ptr64 _TRIAGE_EX_WORK_QUEUE

DOMARS, 2022-07-27:
@Jacksole - Thanks for filing this issue. And @grahamm78 - thanks for seeing the fix here. I updated the doc to use the _TRIAGE_9F_POWER. Good luck with your debugging and thanks again for calling out this error in the docs.

closed 2022-07-27 by DOMARS