MicrosoftDocs/windows-driver-docs #3222: Consider adding a warning that Windows ignores custom friendly names of audio endpoints using the KSNODETYPE_SPEAKER category

asapha, 2022-08-03:
I got stuck for a few hours trying to find a way to change the display/friendly name of my custom audio device until I found [this random post](https://community.osr.com/discussion/comment/304460/#Comment_304460).

> Windows will ignore any friendly name customization for audio endpoints that have a KSPIN_DESCRIPTOR Category that looks like speakers. My guess is there is some backwards compatibility issue involved.

In short, if you have a pin with the `KSNODETYPE_SPEAKER` category:

```c
   {
    0,
    0,
    0,                                              
    NULL,                                           
    {                                               
      0,                                            
      NULL,                                         
      0,                                            
      NULL,                                         
      SIZEOF_ARRAY(PinDataRangePointersBridge),     
      PinDataRangePointersBridge,                   
      KSPIN_DATAFLOW_OUT,                           
      KSPIN_COMMUNICATION_NONE,            
      
      // Category        
      &KSNODETYPE_SPEAKER,

      // The GUID for your custom name/category.                           
      &NAME_MSVAD_SIMPLE,                                         
      0                                             
    }
```

The audio device will always be displayed as "Speakers". (Unless the user changes it manually)

But with `KSNODETYPE_HEADPHONES` for example:
```c
  {
     [...]
      
      // Category                    
      &KSNODETYPE_HEADPHONES,
      
      // The GUID for your custom name/category.
      &NAME_MSVAD_SIMPLE,                                         
      0                                             
    }
```

The custom name will correctly show up in Windows' GUI.

Reproduced on a Windows 11 VM in Hyper-V with the `simple` project from the `msvad` (Windows 8.1) sample.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e33dcab2-c866-a002-5009-244dc3d0b876
* Version Independent ID: 31b04b82-f1f9-3504-eda3-2761f15f96ec
* Content: [Friendly Names for Audio Endpoint Devices - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/friendly-names-for-audio-endpoint-devices)
* Content Source: [windows-driver-docs-pr/audio/friendly-names-for-audio-endpoint-devices.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/friendly-names-for-audio-endpoint-devices.md)
* Product: **windows-hardware**
* Technology: **audio**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2022-08-31:
@asapha - Thanks for this information. We are working on updates to this topic to follow the latest guidance on the use of the registry, which will provide updated INF files. When we are doing that work, I will look at the situation with the KSNODETYPE_SPEAKER naming and update the docs to clarify speaker node naming.

At least two things to look at to determine the final behavior that we will document are, which version of Windows will this apply to, and what the values of all of the associated KSProperties - as there a number that may impact what is rendered in the UI. It is often a combination of KSProperties that will need to be used.

As a starting point, I normally look at the latest sysvad sample, to see if the sample INF file shows an example of the desired behavior.

https://github.com/microsoft/Windows-driver-samples/blob/main/audio/sysvad/TabletAudioSample/tabletaudiosample.inx

I will also check and see if this text from https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/friendly-names-for-audio-endpoint-devices needs updating with the suggested warning.

*Audio devices that expose pins with standard categories GUIDs should include / needs the inbox KS.INF or KSCAPTUR.INF name registration in your device INF. These inbox INFs contain default friendly name definitions for pre-defined category GUIDs that your driver may wish to populate. These are the same GUIDs found in the KsPinDescriptor.Category member of the PCPIN_DESCRIPTOR structure. For example, the category GUID KSNODETYPE_MICROPHONE entry has the associated friendly name "microphone" and the category GUID KSNODETYPE_SPEAKER entry has the associated friendly name "speakers," and so on.*

Thanks again for passing on the behavior you are seeing with a KS node type of speakers.

closed 2022-08-31 by DOMARS

sophy228, 2023-01-18:
@DOMARS Any Update? I have got the same problem and found the same conclusion after 1 day debuging ..

When I am just going to post a feedback, I find this post....

Is it exepected behavior or something needed to be noticed in https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/friendly-names-for-audio-endpoint-devices needs  ?