MicrosoftDocs/windows-driver-docs #3230: WinUSB -> Microcontroller

KahnSoft, 2022-08-12:
Hi,
missing is the other side of communication, how the io pins on microcontroller correlate with the USB Requests. 
This transfer protocoll is unclear. An Microcontroller example is the grant to use the winusb.

mhopkins-msft, 2022-08-12:
Hi @KahnSoft

Can you please tell me which specific documentation page you're referring to?

Thanks,
 ~ *Mark*

KahnSoft, 2022-08-13:
Hi Mark,

i mean that the explain of WinUsb 
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/using-winusb-api-to-communicate-with-a-usb-device

let see only the Windows side, to use the WinUsb  it require an 
Microcontroller,  there must answer the WinUsb Requests (usb-yellow-books) .
This way is close for normal devloopers, no one can use WinUsb without 
the answerprocess, there is not transparent, so all use linux .

We must build the communication to own usb client (Atiny85) with the MS  
HID Device support, to communicate with my chip, an long jury of
reserch and investifation must be start,  why we find no solutions to 
communicate with any hardware via USB. Take an controller put a yellow 
bock request protokoll into to let all see how the speaking will be work.

Here an very slow Hid solution http://www.flexxvision.de/keyman.html

No Bundle transfer only 2Kb Datarate via Service requests to the Atiny85

sorry for my english
  best regards from Berlin

mhopkins-msft, 2022-08-26:
Hi Karsten,

Thank you for your suggestion for an example of how to respond to WinUSB requests. I've created a work item in our backlog for consideration for a future article.

Do you mean that you want to see example code to implement in your microcontroller to respond to WinUSB requests?

Thank you,
 ~ *Mark*

mhopkins-msft, 2022-08-26:
Hi Karsten,

Since I've created the work item, I'm going to close this GitHub issue. Feel free to add any more information you'd like. I appreciate your feedback. Thank you for helping us make the docs better!

 ~ *Mark*

closed 2022-08-26 by mhopkins-msft

KahnSoft, 2022-08-27:
Hello Thx For Answer in this case,

for my Atiny85  i use an codefragment from V-USB  Library via arduino, so i have got the example to 
communicate via A and B Dataline with  Ten-Diodes, and it handle down the YellowBook USB-Descriptor protokoll.
It was in early and now time Luck to find a solution. The programmer wich more transparence to the USB Descriptor -Protokoll.
for example handle the Descriptor with Interrrupt and Assembler, an other way  wo is missing use an RaspiPi A+B Line on Bundle Transfer to emulte an Webcam FPGA Chip   . 

For now i use an HID.Serdial communication wirh ~2Kb datarate   using Hid.dll

For example the atiny85 kernel

`
/* Name: usbdrvasm.S
 * Project: V-USB, virtual USB port for Atmel's(r) AVR(r) microcontrollers
 * Author: Christian Starkjohann
 * Creation Date: 2007-06-13
 * Tabsize: 4
 * Copyright: (c) 2007 by OBJECTIVE DEVELOPMENT Software GmbH
 * License: GNU GPL v2 (see License.txt), GNU GPL v3 or proprietary (CommercialLicense.txt)
 */

/*
General Description:
This module is the assembler part of the USB driver. This file contains
general code (preprocessor acrobatics and CRC computation) and then includes
the file appropriate for the given clock rate.
*/

#define _VECTOR(N)   __vector_ ## N   /* io.h does not define this for asm */
#define macro   .macro
#define endm    .endm
#define nop2    rjmp    .+0 /* jump to next instruction */

#define __SFR_OFFSET 0      /* used by avr-libc's register definitions */

#include "usbdrv.h"         /* for common defs */

/* register names */
#define x1      r16
#define x2      r17
#define shift   r18
#define cnt     r19
#define x3      r20
#define x4      r21
#define x5		r22
#define bitcnt  x5
#define phase   x4
#define leap    x4

/* Some assembler dependent definitions and declarations: */

#ifdef __IAR_SYSTEMS_ASM__
    extern  usbRxBuf, usbDeviceAddr, usbNewDeviceAddr, usbInputBufOffset
    extern  usbCurrentTok, usbRxLen, usbRxToken, usbTxLen
    extern  usbTxBuf, usbTxStatus1, usbTxStatus3
    public  usbCrc16
    public  usbCrc16Append

    COMMON  INTVEC
#   ifndef USB_INTR_VECTOR
        ORG     INT0_vect
#   else /* USB_INTR_VECTOR */
        ORG     USB_INTR_VECTOR
#       undef   USB_INTR_VECTOR
#   endif /* USB_INTR_VECTOR */
#   define  USB_INTR_VECTOR usbInterruptHandler
    rjmp    USB_INTR_VECTOR
    RSEG    CODE

#else /* __IAR_SYSTEMS_ASM__ */
[... 677 lines cut ...]
;2006-03-06: moved transfer of new address to usbDeviceAddr from C-Code to asm:
;set address only after data packet was sent, not after handshake
    subi    YL, 2           ;[0] Only assign address on data packets, not ACK/NAK in r0
    sbci    YH, 0           ;[1]
    breq    skipAddrAssign  ;[2]
    sts     usbDeviceAddr, x2; if not skipped: SE0 is one cycle longer
skipAddrAssign:
;end of usbDeviceAddress transfer
    ldi     x2, 1<<USB_INTR_PENDING_BIT;[4] int0 occurred during TX -- clear pending flag
    USB_STORE_PENDING(x2)   ;[5]
    ori     x1, USBIDLE     ;[6]
    in      x2, USBDDR      ;[7]
    cbr     x2, USBMASK     ;[8] set both pins to input
    mov     x3, x1          ;[9]
    cbr     x3, USBMASK     ;[10] configure no pullup on both pins
    ldi     x4, 4           ;[11]
se0Delay:
    dec     x4              ;[12] [15] [18] [21]
    brne    se0Delay        ;[13] [16] [19] [22]
    out     USBOUT, x1      ;[23] <-- out J (idle) -- end of SE0 (EOP signal)
    out     USBDDR, x2      ;[24] <-- release bus now
    out     USBOUT, x3      ;[25] <-- ensure no pull-up resistors are active
    rjmp    doReturn

`

KahnSoft, 2022-08-27:
Missing also an Solution , the only source atm is : https://www.obdev.at/products/vusb/index.html
(Sorry Code posting dosent work ^^) 

Thanks and regards from Berlin
   Karsten