MicrosoftDocs/windows-driver-docs #3235: These links do not point to Windows 11 SDK/WDK/EWDK installers - they are version 10

fenderluvr, 2022-08-17:
This link does NOT point to v11 (22H2) of the SDK. It points to v10 (old version):
https://developer.microsoft.com/en-us/windows/downloads/windows-sdk/

This link does NOT point to v11 (22H2) of the WDK. It points to v10 (old version):
https://go.microsoft.com/fwlink/?linkid=2196230

This link does NOT point to v11 (22H2) of the EWDK. It points to v10 (old version):
https://docs.microsoft.com/en-us/legal/windows/hardware/enterprise-wdk-license-2022

These links are invalid, or you don't yet have versions of the SDK/WDK/EWDK for Windows 11. Either correct this page OR delete the links because there are issues (as you denote on this page) with using the these older versions with Visual Studio 2022.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **whats-new-and-downloads**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-08-22:
Hi @fenderluvr thanks for the feedback.  In fact, it was indeed the v11 (22H2) kit, it just so happens that the version number for that kit is `10.0.22621.1`.  However, we just replaced it with an updated version of the v11 (22H2) kit last Friday -- with a new version number of `10.0.22621.382`.  The new kit is available from the same download link as before -- https://go.microsoft.com/fwlink/?linkid=2196230.

For more info about the update, see the Note at the top of https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk .

Thanks again for reaching out!

closed 2022-08-22 by tedhudek

fenderluvr, 2022-08-23:
If true, that's a very confusing versioning method - at best. Also, I
notice that you didn't address the WDK/EWDK in your response but closed out
the request.