MicrosoftDocs/windows-driver-docs #3241: Windows Store link for Windbg preview broken

ElizabethGreene, 2022-08-22:
This isn't a documentation issue, but I'm not sure where else to open an issue for the WinDBG preview.  The web link to the 'WinDBG preview' page on the Microsoft store is https://www.microsoft.com/store/productId/9PGJGD53TN86 , but that page is broken.  It says "Not Available for bots to index".

Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1e093f9-7df0-17ab-71ae-8af9003ee041
* Version Independent ID: 55ed6fd3-a7a6-f7d5-2a8e-fbf6cc6e2c7b
* Content: [Debugging Using WinDbg Preview - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-using-windbg-preview)
* Content Source: [windows-driver-docs-pr/debugger/debugging-using-windbg-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-using-windbg-preview.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2022-08-22:
@ElizabethGreene - That link works for me now. Perhaps try a different Windows PC?

DOMARS, 2022-08-25:
@ElizabethGreene - I hope the Store download is working OK by now.  If not, I would recommend that you file a feedback issue on the debugger WinDbg GitHub feedback site: https://github.com/microsoftfeedback/WinDbg-Feedback/issues
Thanks for letting us know what you ran into with the Windows Store.

closed 2022-08-25 by DOMARS