MicrosoftDocs/windows-driver-docs #3247: Clarification required for the AppId and AppName fields in MBIM_CID_MS_UICC_APP_LIST

aleksander0m, 2022-08-23:
The `MBIM_CID_MS_UICC_APP_LIST` request returns the `MBIM_UICC_APP_LIST` structure in the response, which contains a list of `MBIM_UICC_APP_INFO` structures.

The `MBIM_UICC_APP_INFO` structure defines:
 * the "AppId" as an array of bytes represented by an Offset-Length pair (AppIdOffset and AppIdSize).
 * the "AppName" as a UTF-8 string represented by an Offset-Length pair (AppNameOffset and AppNameLength).
 
In the AppIdOffset documentation says, quoting:

> Offset for the application ID in the databuffer. Only the first AppIdSize bytes are meaningful. If the application ID is longer than MBIM_MAXLENGTH_APPID bytes, then AppIdSize specifies the actual length but only the first MBIM_MAXLENGTH_APPID bytes are in this field.

Does this mean that the if the value given in AppIdSize is greater than MBIM_MAXLENGTH_APPID, the AppIdSize must be ignored and instead only MBIM_MAXLENGTH_APPID bytes should be read?

And if so, why is the AppIdSize format defined as SIZE (0..16) in the table?

In a similar way, the AppNameOffset documentation says:

> The length of this field is specified by AppNameLength. If the length is greater than or equal to MBIM_MAXLENGTH_APPNAME bytes, this field contains the first MBIM_MAXLENGTH_APPNAME - 1 bytes of the name. The string is always null-terminated.

Does this mean that the if the value given in AppNameLength is greater than MBIM_MAXLENGTH_APPNAME bytes, the AppIdLength must be ignored and instead only MBIM_MAXLENGTH_APPNAME bytes should be read?

And if so, why is the AppNameLength format defined as SIZE (0..256) in the table?

E.g. Is it that AppNameLength may be 1024, but because MBIM_MAXLENGTH_APPNAME is 256, the field in the MBIM message will only contain 256 bytes, even if the AppNameLength value is much longer?

I believe this is the first time I've seen this kind of logic/restriction in an OL pair defined in MBIM, so wanted to confirm.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c31681f0-c52b-a074-af00-b63d0ba7c52d
* Version Independent ID: 4b4a4655-afed-9320-a98b-574dceefd742
* Content: [MB UICC application and file system access - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-uicc-application-and-file-system-access)
* Content Source: [windows-driver-docs-pr/network/mb-uicc-application-and-file-system-access.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-uicc-application-and-file-system-access.md)
* Product: **windows-hardware**
* Technology: **network**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2022-08-23:
Hi @aleksander0m,
Thanks for your questions. I've contacted the development team and will let you know when I hear back.

aviviano, 2022-12-21:
Hi @aleksander0m,
Thanks for opening this issue. To answer your questions:
"Does this mean that the if the value given in AppIdSize is greater than MBIM_MAXLENGTH_APPID, the AppIdSize must be ignored and instead only MBIM_MAXLENGTH_APPID bytes should be read?" 
**Yes, you are correct.**
"And if so, why is the AppIdSize format defined as SIZE (0..16) in the table?"
**AppIdSize may contain a number greater than 16, but in this case only the first 16 (MBIM_MAXLENGTH_APPID) bytes are in the databuffer.**
Similarly, your understanding of AppNameOffset/AppNameLength is correct.
We are considering clarifying these points in the documentation and have created an internal work item. I'll close this issue now, but if you have further comments/questions please open a new issue.
Thanks again,
Amy

closed 2022-12-21 by aviviano

aleksander0m, 2022-12-21:
>> "Does this mean that the if the value given in AppIdSize is greater than MBIM_MAXLENGTH_APPID, the AppIdSize must be ignored and instead only MBIM_MAXLENGTH_APPID bytes should be read?"
>Yes, you are correct.

Thanks for confirming and reviewing this, really appreciate it

For future reference, these are the changes going into libmbim in order to support this special behavior which applies exclusively to the MBIM_CID_MS_UICC_APP_LIST message: https://gitlab.freedesktop.org/mobile-broadband/libmbim/-/merge_requests/155