MicrosoftDocs/windows-driver-docs #3248: Parameter 1 has changed

rodwiddowson, 2022-08-24:
It seems to me that Parameter 1 is no longer (The calling address in the driver that locked the pages* but rather the pointer to a **MmLk** structure.  This contains, inter alia, a calling stack, starting at offset 0x48 from the supplied address:
```

Arg1: ffff8b8c4dc1c520, The calling address in the driver that locked the pages or if the
	IO manager locked the pages this points to the dispatch routine of
	the top driver on the stack to which the IRP was sent.

3: kd> !pool ffff8b8c4dc1c520 2
Pool page ffff8b8c4dc1c520 region is Nonpaged pool
*ffff8b8c4dc1c510 size:   a0 previous size:    0  (Allocated) *MmLk
		Owning component : Unknown (update pooltag.txt)

3: kd> dq ffff8b8c4dc1c520
ffff8b8c`4dc1c520  00000000`00000000 00000000`00000000
ffff8b8c`4dc1c530  ffff8b8c`4bc0eba0 ffff8b8c`47a74880
ffff8b8c`4dc1c540  00007df4`faa21000 00000000`00000001
ffff8b8c`4dc1c550  00000200`00000000 037a882f`00000003
ffff8b8c`4dc1c560  00000000`0001c0ff fffff801`80d4a6f1
ffff8b8c`4dc1c570  fffff801`80c808bd fffff801`80b0e619
ffff8b8c`4dc1c580  fffff801`80e9a1fc fffff801`81b331de
ffff8b8c`4dc1c590  fffff801`81b39988 fffff801`81b2bf56

3: kd>  ln fffff801`80d4a6f1
(fffff801`80d4a5f4)   nt!MiAddMdlTracker+0xfd   |  (fffff801`80d4a81c)   nt!MiDeliverPicoExceptionForProbedPage

3: kd> ln fffff801`80c808bd 
(fffff801`80b0e630)   nt!MiProbeAndLockPages+0x17228d
```
I could make up some words and put in a PR but this feels like something that an appropriate dev would be better placed to supply.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 674f701c-05fa-9310-a183-69de98471c88
* Version Independent ID: f3d33b57-40a6-5d31-b503-7944e7ba410f
* Content: [Bug Check 0xCB DRIVER_LEFT_LOCKED_PAGES_IN_PROCESS - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xcb--driver-left-locked-pages-in-process)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xcb--driver-left-locked-pages-in-process.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xcb--driver-left-locked-pages-in-process.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2022-08-31:
@rodwiddowson - Thanks, yes it looks like things have changed here. I'm also seeing that param 2 is always zero (reserved). I will follow up with the dev team, and see if we can also update !analyze as well.

DOMARS, 2022-09-09:
@rodwiddowson - I have confirmed with the dev team that this has changed. We are working on descriptions and other clarifications on the three parameters.

DOMARS, 2022-10-05:
@rodwiddowson - Topic is updated with the new parameters. I also added a link to the [!lockedpages](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/-lockedpages) debugger extension, which should display all of the locked MDLs for the current process. Thanks again for this issue, and your help in improving the docs for the next person down the line.

closed 2022-10-05 by DOMARS

rodwiddowson, 2022-10-05:
@DOMARS Thank **you** (and your colleagues).  It is a pleasure to be allowed to help nudge the qualify of the documentation up even further