MicrosoftDocs/windows-driver-docs #3250: Request more specific guide of logging with DSS for MBB driver vendor and WinRT App

forecs, 2022-08-31:
[Enter feedback here]

At the beginning of this document, it says:
> If you are planning for your modem to support MBIM_CID_MODEM_LOGGING_CONFIG, please provide feedback on this page so that we can best support you. This CID is currently experimental and has not been tested with a modem yet, as none support it.

My customer, as one 4G/5G MB device vendor, are interested to support MB modem logging with DSS on their MB driver.
After reading this document and searching the Internet, it seems not much specific guide on this topic, would like to know: 
 * Is there any sample driver code for reference?
 * Any diff which may take care of between 4G and 5G MB device driver?
 * Is there any sample WinRT code for reference?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9a49c708-4a65-da81-5e4a-79887a5082d5
* Version Independent ID: b59ce8c9-0a81-f15e-2991-e7f66cdb0648
* Content: [MB modem logging with DSS - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-modem-logging-with-dss)
* Content Source: [windows-driver-docs-pr/network/mb-modem-logging-with-dss.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-modem-logging-with-dss.md)
* Product: **windows-hardware**
* Technology: **network**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2022-08-31:
Thanks for opening this issue @forecs. I've reached out to the development team with your questions and will let you know when I hear back.

aviviano, 2022-08-31:
Hi @forecs,
Closing this issue as the discussion is on going via email.

closed 2022-08-31 by aviviano