MicrosoftDocs/windows-driver-docs #3255: FilePathOffset documentation in MBIM_UICC_FILE_PATH is a bit misleading

aleksander0m, 2022-09-06:
The documentation for the `FilePathOffset` field in the `MBIM_UICC_FILE_PATH` structure says, quoting:

> The file path is an array of 16-bit file IDs. The first ID must be either 0x7FFF or 0x3F00. If the first ID is 0x7FFF, then the path is relative to the ADF of the application desginated by AppId. Otherwise, it is an absolute path starting from the MF.

And then, the `FilePathSize` says, quoting:

> The size of the file path, in bytes.

If the size field is in bytes, in MBIM terms that would have meant that the file path array is a byte array (the size fields exposes number of items in the array), and that byte array happens to contain a series of 16-bit fields encoded in big endian format. I would not refer to it as an array of 16-bit fields, because in MBIM all integer fields are little endian encoded, so that adds to the confusion.

E.g. the file path field may contain an array of bytes as follows to refer to the EF_GID1 file: `7FFF6F3E`.

So my suggestion is to rewrite the  `FilePathOffset` field documentation so that it clearly says it is a **byte array** that contains **a series of 16-bit numbers encoded in big endian format**. For reference, that is exactly how the Qualcomm SDX55 module implements this method (testing here with ' mbimcli`):

```
$ sudo mbimcli -p -d /dev/wwan0mbim0 --ms-query-uicc-file-status="application-id=A0000000871002FF34FF0789312E30FF,file-path=7FFF6F3E"
...
[06 sep 2022, 22:33:15] [Debug] [/dev/wwan0mbim0] sent message (translated)...
<<<<<< Header:
<<<<<<   length      = 88
<<<<<<   type        = command (0x00000003)
<<<<<<   transaction = 3
<<<<<< Fragment header:
<<<<<<   total   = 1
<<<<<<   current = 0
<<<<<< Contents:
<<<<<<   service = 'ms-uicc-low-level-access' (c2f6588e-f037-4bc9-8665-f4d44bd09367)
<<<<<<   cid     = 'file-status' (0x00000008)
<<<<<<   type    = 'query' (0x00000000)
<<<<<< Fields:
<<<<<<   Version = '1'
<<<<<<   ApplicationId = 'a0:00:00:00:87:10:02:ff:34:ff:07:89:31:2e:30:ff'
<<<<<<   FilePath = '7f:ff:6f:3e'
...
[/dev/wwan0mbim0] UICC file status retrieved:
	    Status word 1: 0x00000090
	    Status word 2: 0x00000000
	    Accessibility: unknown
	             Type: unknown
	        Structure: transparent
	       Item count: 1
	        Item size: 1
	Access conditions:
		      Read: pin1
		    Update: adm
		  Activate: adm
		Deactivate: adm

```

Let me know what do you think. 

Cheers!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c31681f0-c52b-a074-af00-b63d0ba7c52d
* Version Independent ID: 4b4a4655-afed-9320-a98b-574dceefd742
* Content: [MB UICC application and file system access - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-uicc-application-and-file-system-access)
* Content Source: [windows-driver-docs-pr/network/mb-uicc-application-and-file-system-access.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-uicc-application-and-file-system-access.md)
* Product: **windows-hardware**
* Technology: **network**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2022-09-06:
Thanks for your suggestion @aleksander0m. I'll pass this on to the development team and let you know when I hear back.

aviviano, 2022-12-20:
Hi @aleksander0m,
Thank you so much for bringing this to our attention. Your understanding, and the Qualcomm SDX55 module’s implementation, are correct. Microsoft generally reviewed the documentation and definitions with modem IHVs such as Qualcomm during the development of the CID. Microsoft co-engineered with major modem IHVs such as Qualcomm during their first implementation of the CID. As such, we feel confident that major modem IHVs understands this CID correctly and implemented it as expected. The ambiguity of the definition on this point is unfortunate and is a mistake on Microsoft side. But at this point, we do not feel it is necessary to change the definition and documentation.

closed 2022-12-20 by aviviano

aleksander0m, 2022-12-21:
> The ambiguity of the definition on this point is unfortunate and is a mistake on Microsoft side. But at this point, we do not feel it is necessary to change the definition and documentation.

Ok, thanks. It would have been a simple change and it would have clarified the documentation, and wouldn't have required any change of code anywhere, but well :) Thanks for looking at this.