MicrosoftDocs/windows-driver-docs #3260: pnp.exe

Li77-22oR, 2022-09-11:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 18777b73-5898-9e38-c074-5d5ec1572d80
* Version Independent ID: d08c4303-6dee-049a-1c69-620b5bcb68f9
* Content: [PnPUtil - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil)
* Content Source: [windows-driver-docs-pr/devtest/pnputil.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/pnputil.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

Li77-22oR, 2022-09-11:
> [Enter feedback here]

GitHub just want say i star LOVE you

DOMARS, 2022-09-12:
@Li77-22oR, thanks for reaching out. I don't see a question or doc issue, so I'm going to close this out.

closed 2022-09-12 by DOMARS