MicrosoftDocs/windows-driver-docs #3265: Previous Windows documentation

cyberluke, 2022-09-14:
Hi, where to get Windows Kernel Driver documentation for:

- Windows 8.1 (WDK)
- Windows 7 (DDK)
- Windows XP (DDK)
- Windows 98 SE (DDK)

tedhudek, 2022-09-16:
Hi @cyberluke thanks for the question.  The current docs on the web are in fact a superset of all the older documentation.  We add new information but old information stays unless something is deprecated.  When an API is deprecated, we either mark the page as deprecated and leave it in place, or, in some (rare) cases, we move it to https://docs.microsoft.com/en-us/previous-versions/windows/hardware/.

If you want to see how a specific page looked at a specific date, you can also try the [Internet Archive Wayback Machine](https://web.archive.org/).

Hope this helps!

cyberluke, 2022-09-17:
@tedhudek  That is not true.

DDK vs WDK = totally different operation

I ask for specific stuff here:
Windows 8.1 (WDK)
Windows 7 (DDK)
Windows XP (DDK)
Windows 98 SE (DDK)

Where to find them?

Your answer does not cover how to find them on Archive.org. Specific date is very unspecific answer.

cyberluke, 2022-09-17:
Current documentation is focused on Windows 11 only and does not apply for Windows 7, for example. Windows XP is not there at all.

I did expect you would point me at least here on Github MicrosoftDocs! Why you don't have lets say branch for WinXP, Win7, Win8, Win98, where people could access the complete documentation? Current documentation process is rewriting, deleting and obfuscating access to information. 

Therefore I strongly object against current state and please can you help me?

cyberluke, 2022-09-17:
Interface change, parameters change. Incompatible parameters. Different classes. Different directory structure across all Windows SDK versions.

cyberluke, 2022-09-17:
Look here:

Sometimes you have a link with version at the end **(v=vs.85)**: https://learn.microsoft.com/en-us/previous-versions/windows/desktop/ee417917(v=vs.85)

How to find such revisions? Do they have some entry point URL?

cyberluke, 2022-09-18:
Where is this link?

Wrong redirect.

 http://www.microsoft.com/whdc/devtools/wdk/wdkdocs.mspx

cyberluke, 2022-09-18:
@tedhudek  SDK archive contains only Win7: https://developer.microsoft.com/en-us/windows/downloads/sdk-archive/

cyberluke, 2022-09-18:
please do the homework once more and reopen this issue

tedhudek, 2022-09-19:
Hi @cyberluke,

What is the specific problem you are trying to solve?

Without knowing that, here are a couple ideas:

You could try installing previous kits from https://learn.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads#step-2-install-the-wdk.

You could also try cloning https://github.com/MicrosoftDocs/windows-driver-docs and https://github.com/MicrosoftDocs/windows-driver-docs-ddi to your local and searching for OS names.

You could ask on [OSR's NTDEV forum](https://community.osr.com/categories/ntdev) or [windows-hardware-wdk](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk.html).  

Sorry, I wish I could help more, but we don’t keep a public archive of content for platforms we no longer support.

vxiiduu, 2022-09-20:
> Hi, where to get Windows Kernel Driver documentation for:
> 
> * Windows 8.1 (WDK)
> * Windows 7 (DDK)
> * Windows XP (DDK)
> * Windows 98 SE (DDK)

The official documentation for basically everything Windows-related is included in visual studio. For example, this picture is what I have in the "offline help" program that comes with Visual Studio 2010, which mostly covers Windows 7 but has a few articles talking about Windows 8 as well. I'm sure you can find VS2008 and earlier if you want specific information about Vista, XP etc. (although for XP specifically you can just read the source code because it's leaked).

[image]

cyberluke, 2022-09-23:
Ok, so MSDN subscription would not provide any advantage for having access to even offline documentation for WDK/SDK older than Windows 7?

I need to recompile a few open-source drivers (such as vJoy) that have Win XP support, for example. Then also backport some stuff in near future. It will be shared with retrogaming community as well as for museums. And it is interesting for younger generation to see what they would have to do in the past to run some game. As it was not always one click play stuff and you had to understand IRQ and DMA. In Europe, on specialized technical high school, they still teach IRQ, DMA and x86 hardware basics. So it is also something that should be preserved as a part of human history.

I bought a bundle of 20 random CDs from E-Bay containing various WDK, SDK and MSDN CDs as well as boxed Visual Studio (older versions) - for collector purposes.

My dev room consists of eight computers with KVM. My main development machine has multiboot with Win10, Win7, WinXP. The main motivation was to have one place for WDK documentation including search feature. Because WDK before Win7 was not integrated in Visual Studio, but offers a separate compiler and separate command line environment.
[image]

I was hoping this documentation would be added here on MicrosoftDocs. Even it is unsupported, but Microsoft supports open-source now more than 20 years ago. Therefore it is unlucky that information from website is being removed or rewritten. I understand that, it is good for customer experience. But I was hoping there would be some pre-Win7 documentation somewhere in this repository. It would be nice, if it could be added in some separate branch, only on Github.

tedhudek, 2022-09-23:
Hey Lukas, that's so cool - I showed your message to the team of writers who own the WDK docs.  Love your mission.

Actually, I think you've hit upon your best option, namely CDs containing the old kits.  I can't speak to the MSDN subscription idea, because I've never looked into what you get there.  But the forums (see links above) might give you a way to connect with people who would know about that.  There are many long-timers in the driver community.  In particular, Peter Viscarola, who runs the OSR forum, could probably tell you about that.  But for the CDs specifically, pre-Windows 7, when it was still called the DDK, you could get a CD with the DDK for Windows XP and there was one for the DDK for Windows 2000.  I'm not sure if the docs are on those CDs, but it might be.  Again, the forums might help you find people who are also interested in these older versions.

If all that fails, the superset model really is true.  I know you would prefer to see the old versions without all the new content layered on top, but the old stuff really is still there.  It's like a very old house that has a (couple) totally new shell(s) built around it.

tedhudek, 2022-10-19:
How did it turn out, Lukas?  Were you able to find Windows Kernel Driver documentation for Win98/XP/7/8.1?

Also - thanks to @vxiiduu for the reply above about the Visual Studio documentation viewer.

cyberluke, 2022-10-21:
You gave me only 3 days to respond during a busy week before closing the ticket.

tedhudek, 2022-10-21:
Happy to reopen it, Lukas.  What's up?

cyberluke, 2022-10-21:
No was not able to find. For example in WinDDK 7600, there is only folder Help and it contains some file with .hxi suffix. What is that? Cannot be opened. 

Next, there must be some step by step tutorial, some book or best practices and troubleshooting. I think example projects not enough.

cyberluke, 2022-10-21:
To be exact, here is a screenshot of WinDDK 7600 after install on the left and WinXP DDK ISO CD on the right.

There is nothing on the right. On the left only Help folder full of *.hxi files.

[image]

cyberluke, 2022-10-21:
To be fair, Win98 contains at least this: WIN98DDK.CAB, which contains HELP_WIN98DDK.CHM and can be opened, but it seems to be an exception.

tedhudek, 2022-10-21:
The hxs/hxi was an old file format that required a separate viewer, not sure if you can get one now.  Maybe if you install the kit it includes a viewer?  Glad you found a chm for Win98.

Did you try asking on the forums?

cyberluke, 2022-10-21:
But also Win98 CHM file says this:
"The Essential Windows 95/98 DDK Resource Database is a large database containing pointers to DDK resources supplied by Microsoft and third parties."

These pointers will be some sort of invalid website links, I guess.

cyberluke, 2022-10-21:
HXI: Google find Super File Viewer with this Microsoft Store link: https://apps.microsoft.com/store/detail/super-file-viewer/9N5NM3V59677

but it gets redirected to Store home page.

cyberluke, 2022-10-21:
I have installed the kit here c:\WinDDK\7600.16385.1\help\ - that is Windows 7 DDK installed in Windows 10 - not that old.

cyberluke, 2022-10-21:
Well, I did not get much help here yet :-) most of the time I'm the one, who need to investigate and search online and then come back here. It is still far from getting some table with all Windows DDK versions and list of files or website links containing the help.

cyberluke, 2022-10-21:
So:
Win98 : contains CHM
WinXP: cannot find anything yet
Win7: contains HXI
Win8.1: contains only URL link??? 

Win8.1 issue:
WindowsDriverKitOnlineDocumentation.url contains link to http://go.microsoft.com/fwlink/?LinkId=301383 which redirects to https://learn.microsoft.com/en-us/windows-hardware/drivers/?redirectedfrom=MSDN

That will be a problem in the future! Win8.1 does not contain any documentation inside Help directory as Win7 did. What it means? Now the website redirects to Windows 11 Hardware development help. I suspect in 5 years it will get even more obsolete and valid Win8 information will be unreachable.

cyberluke, 2022-10-21:
Win10 DDK points to same URL as Win8.1: http://go.microsoft.com/fwlink/?LinkId=301383 ....lets be honest, you will remove Win8.1 information from that website as soon as you can. This happened to WinXP and it started happening to Win7. So one day, it will be gone for Win8 as well.

cyberluke, 2022-10-21:
Here, Win8.1 still "temporarily" unavailable: https://learn.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads

No mention of Win7 DDK.

There is mention of Win8 using MSBuild. But then zero information. It is cluttered all over the place. If you search online documentation of "MSBuild", you will get these links from different categories:
1) https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/wdk-tasks-for-msbuild
2) https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/-static-driver-verifier-commands--msbuild-
3) https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/msbuild-primer-for-wdk-developers
(84 results total)

So it is very difficult to gain knowledge specific to some Windows version. 

It forces everyone to download Windows 11 and use the latest Visual Studio and choose the easy route for businesses.

cyberluke, 2022-10-21:
At this point, I would prefer maybe some book for each Windows version WDK and DDK by O'Reilly or someone. Would you know if there exists some good knowledge resource for, lets say, newcomers to the subject?

I don't care how much it will cost because I need to save my time as well. I would even pay extra money for Microsoft MSDN subscription, if they would have streamlined specific Windows version documentation and help for developers.

tedhudek, 2022-10-25:
Hi Lukas,

I empathize with you, and I’m so very sorry that we’re not able to help you here.

We just don’t have these doc sets for systems no longer supported. The bottom line remains that we do not have what you’re asking for.

Here's the best we could find on the web, maybe there's something here you haven't seen?

- https://archive.org/details/tool-microsoft-ddk-wdk
- https://www.microsoft.com/en-us/download/details.aspx?id=11800
- https://winworldpc.com/product/windows-sdk-ddk/xp-nt-51
- https://msfn.org/board/topic/101180-windows-98-ddk/
 
Again, I think the forums (links above) are a possible venue where you can find a solution.  There are lots of like-minded people there who are also interested in the docs from years past, and perhaps some of them have set aside their own collections of the old docs.

Wishing you all the best in your search.

closed 2022-10-25 by tedhudek

cyberluke, 2022-10-25:
Thank you, but I’m the one who created a similar topics on MSFN forums already before going there. They did not help and then I start to seek help here. So we have a little bit of circular dependency going on.

Thanks for a short sum up, perhaps it will help some other readers. I need to find some book if it exists…