MicrosoftDocs/windows-driver-docs #3268: Update the IRQ_DEVICE_POLICY table to include all members

msudhakargit, 2022-09-15:
Update the IRQ_DEVICE_POLICY table to include all members i.e. IrqPolicyAllProcessorsInMachineWhenSteered as well.
Add notes when appropriate on how these perform on heterogenous/hybrid systems, i.e. when a system has Performance-Cores (P-Cores) and Efficient Cores (E-Cores)

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 63a9c4b1-f36a-e0ca-6951-7f6bd4228b43
* Version Independent ID: b5c0fac6-b184-b553-9f0b-0bb6f06caec4
* Content: [Interrupt Affinity - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/interrupt-affinity-and-priority)
* Content Source: [windows-driver-docs-pr/kernel/interrupt-affinity-and-priority.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/interrupt-affinity-and-priority.md)
* Product: **windows-hardware**
* Technology: **kernel**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-09-15:
Thanks for the feedback @msudhakargit.  It looks like https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_irq_device_policy has info about `IrqPolicyAllProcessorsInMachineWhenSteered`. I will see if I can find any more info that we could add to https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/interrupt-affinity-and-priority.

tedhudek, 2022-09-20:
We've made a couple updates.

First, on https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/interrupt-affinity-and-priority:

* Added recommendation to use default value:
[image]
* Removed the table, since it was duplicated (and more recent) in https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_irq_device_policy. Linked to that page instead.

Second, on https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_irq_device_policy:

* Updated to indicate that `IrqPolicyAllProcessorsInMachineWhenSteered` is reserved for system use.

Thanks for the feedback!

closed 2022-09-20 by tedhudek

msudhakargit, 2022-09-20:
Thanks Ted!