MicrosoftDocs/windows-driver-docs #3288: The "Win32Thread" may not mean "system service dispatch table", is it wrong?

ccteam, 2022-10-09:
[Enter feedback here]
The "Win32Thread" may not mean "system service dispatch table", is it wrong?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 8dffb15e-1398-c43f-9c6d-4da7b0f6d4f2
* Version Independent ID: 15db6baa-e051-546b-cadc-e3ceaaed1546
* Content: [thread (WinDbg) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/-thread)
* Content Source: [windows-driver-docs-pr/debugger/-thread.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-thread.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2022-10-21:
@ccteam - Yes, the entry in the table was wrong and I corrected it. As you might have guessed, the provided address is a pointer to Win32Thread. Thanks for seeing this error and filing this issue. Good luck with your kernel thread debugging work.

closed 2022-10-21 by DOMARS

ccteam, 2022-12-10:
Thanks for confirmation.