MicrosoftDocs/windows-driver-docs #3290: How to disable this GPO?

mct370770, 2022-10-10:
[Enter feedback here]
As stated in this document - when you receive this error 
"This device is blocked from starting while the user is not logged in. (Code 55)"
Then it is because of "the device is blocked from enumeration due to a Group Policy"

It would be most beneficial for customers to know how to disable this Group Policy.
Very many customers has hardware that currently is worth nothing because of this.

Unless you want each and every customer who are using thunderbolt connected devices (external disk arrays, external graphic cards etc) to start using MacOS - because here it works!

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: c6744b3e-ee2f-c905-7a0f-463f5a375cd8
* Version Independent ID: 08b86303-c41c-723a-b70d-43ba29aacfe6
* Content: [CM_PROB_CONSOLE_LOCKED - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/cm-prob-console-locked)
* Content Source: [windows-driver-docs-pr/install/cm-prob-console-locked.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/cm-prob-console-locked.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-10-10:
Hi @mct370770, thanks for the question.

Can you please elaborate on your scenario a bit more? Are these Thunderbolt devices and is the user logged into the computer?  Or are you using remote desktop or some other method?

mct370770, 2022-10-10:
I have a bunch of Intel NUC10 computers, some are NUC10i7FNH and some are NUC10i7FNK. Additionally I have some Mac Mini 2018. They are all equipped with 64Gbyte DRAM and 2Tbyte NVMe. Externally I have some Lacie 6Big 60Tbyte arrays and some Razor Core X with AMD RTX3070 cards. These devices are shared between the computers - and they are all connected through thunderbolt 3. 

When I boot a computer with a device already attached there are never any problems.

but if I decide after boot to attach one of the thunderbolt 3 devices then on Windows I run into that specific problem - the problem never shows in MacOS on a Mac Mini - but it does show on the Mac Mini if I boot it with Windows.

mct370770, 2022-10-18:
As stated - when I attach a thunderbolt device to a Windows based computer after boot and after login, then the device is unaccessable - and the device driver is saying: 
"This device is blocked from starting while the user is not logged in. (Code 55)"

And when searching for the RCA for this, then you will be directed to Microsoft saying this:
Code 55 - CM_PROB_CONSOLE_LOCKED
This Device Manager error message indicates that the device is blocked from enumeration due to a Group Policy / MDM (Intune) policy to protect from DMA attacks.

and I would like to know where to find this Group Policy - and more - how to disable it in this scenario.

tedhudek, 2022-10-19:
Hi Michael,

The relevant group policy is documented on https://learn.microsoft.com/en-us/windows/client-management/mdm/policy-csp-dmaguard.

You can use the Group Policy Editor UI to change the setting there if you need to.

I went ahead and added a link to the above page to the page you were on, https://learn.microsoft.com/en-us/windows-hardware/drivers/install/cm-prob-console-locked.

Hope this helps!

Ted

closed 2022-10-19 by tedhudek