MicrosoftDocs/windows-driver-docs #3297: Add Bluetooth LE device interface GUIDs

DJm00n, 2022-10-14:
There is no information about these device interfaces that are defined in [bthledef.h](https://learn.microsoft.com/windows/win32/api/bthledef/) header since Windows 8:
 - `GUID_BLUETOOTH_GATT_SERVICE_DEVICE_INTERFACE` {781aee18-7733-4ce4-add0-91f41c67b592} - Bluetooth LE device interface GUID
- `GUID_BLUETOOTHLE_DEVICE_INTERFACE` {6e3bb679-4372-40c8-9eaa-4509df260cd8} - Bluetooth LE Service device interface GUID

https://learn.microsoft.com/en-us/windows/win32/api/_bltooth/

GrantMeStrength, 2022-10-20:
Steve - could you please take a look? Thanks!

stevewhims, 2022-10-21:
Thanks, @DJm00n. I've documented the [IsBthLEUuidMatch function](https://learn.microsoft.com/windows/win32/api/bthledef/nf-bthledef-isbthleuuidmatch).

As for how/where to document the GUIDs, I'm investigating.

Thanks!
-Steve

closed 2022-10-21 by stevewhims