MicrosoftDocs/windows-driver-docs #3318: カーネルモード署名 は現在も利用可能なのか？

hiusui, 2022-11-07:
[Enter feedback here]

弊社ではGlobalSignのEVコードサイニング証明書を使用しているのですが、
カーネルモード署名用クロスルート証明書について、

引き続き2021年4月15日以降もカーネルモード署名をご利用いただく場合は、
お手数ですがWindowsデベロッパーセンターにてユーザ登録のうえ、
マイクロソフト社への依頼を行ってください。
https://jp.globalsign.com/support/codesign/manual/win-kernel-mode.html

とありました。

ただ、御社のドキュメントを読む限り、
カーネルモード署名が使用できるようには読めませんでした。

実際にところどうなのかご教示願います。

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 0e326d3e-833a-e312-d35d-11569690c7e6
* Version Independent ID: 7a9795cf-093e-a67f-8162-b832e1161ba8
* Content: [Deprecation of Software Publisher Certificates, Commercial Release and Test Certificates - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates)
* Content Source: [windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-11-07:
Translated:

We use GlobalSign's EV code signing certificate.
About cross-root certificates for kernel-mode signing

If you want to continue using kernel mode signing after April 15, 2021,
We apologize for the inconvenience, but after registering as a user at the Windows Developer Center,
Please make a request to Microsoft.
https://jp.globalsign.com/support/codesign/manual/win-kernel-mode.html

There it was.

However, as far as reading your documentation,
Kernel-mode signatures could not be read for use.

Please tell us how it actually is.

tedhudek, 2022-11-07:
Hello @hiusui, I'm sorry but I can only help you in English.  What version of Windows are you targeting?

hiusui, 2022-11-15:
Sorry, my mistake did not result in a proper inquiry.
Therefore, I will withdraw this inquiry.

closed 2022-11-15 by hiusui