MicrosoftDocs/windows-driver-docs #3323: Where are the DPInst.exe return codes?

FunDeckHermit, 2022-11-10:
I had to use [the wayback machine](https://web.archive.org/web/20130308020906/http://msdn.microsoft.com/en-us/library/windows/hardware/ff544790(v=vs.85).aspx) to get these. Please include them somewhere on the website.

[image]

---
#### Documentdetails

⚠ *Dit gedeelte niet bewerken. Het is vereist voor de koppeling learn.microsoft.com ➟ GitHub-probleem.*

* ID: c7fbbc46-ac1e-3d04-e9b6-a925c28cba71
* Version Independent ID: a9e3d658-8a8e-0f09-af51-5f64427ea3be
* Content: [DIFx Guidelines - Windows drivers](https://learn.microsoft.com/nl-nl/windows-hardware/drivers/install/difx-guidelines)
* Content Source: [windows-driver-docs-pr/install/difx-guidelines.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/difx-guidelines.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-11-22:
Hi Roy, thanks for the report and sorry it's taken a while to get back to you.  Can you tell me more about your scenario?  As the DIFx Guidelines page mentions, DIFx has been deprecated since Windows 10, version 1607.  That page is currently the only doc page we have for it.

And yes, the Wayback Machine is amazing!  I like it too.

FunDeckHermit, 2022-11-23:
We create a product that uses FTDI hardware and FTDI drivers. These drivers can be installed with a simple installer from the [FTDI website](https://ftdichip.com/drivers/d2xx-drivers/). This simple installer is nothing more then dpinst.exe and two drivers packaged in a self-extracting archive installer. 

To package these drivers into our software (Wix Installer) we extracted the contents and made it silent by tweaking the dpinst.exe command lines. 

We cannot rely on Windows Update driver search as systems are often air-gapped and the FTDI installer cannot be made silent.

tedhudek, 2022-11-30:
Thanks Roy!  Appreciate your feedback.  I've added that page back to the docs.  You can find it at https://learn.microsoft.com/en-us/windows-hardware/drivers/install/dpinst-return-code.

closed 2022-11-30 by tedhudek