MicrosoftDocs/windows-driver-docs #3325: when publishToWindows10s how do i chose the security measures and driver requirements with API

HaraldWang, 2022-11-10:
[Enter feedback here]
when i use this API:https://manage.devcenter.microsoft.com/v2.0/my/hardware/products/{productID}/submissions/{submissionId}/shippingLabels
to create a shippinglabel, and this label support windows 10s. when i operater it in website, it must checked with the two option:
[image]

but when i use API to achieve these two option , i have only find the first one like that:
[image]
 
and the second checkbox appears when the first one is checked. and i can not find the second one with API accomplish.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 6b72ba40-cd97-6f54-ec86-b7369b5db3de
* Version Independent ID: c52a67f0-9e83-9bb3-7811-4355e719d9a2
* Content: [Create a new shipping label - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-shipping-label)
* Content Source: [windows-driver-docs-pr/dashboard/create-a-new-shipping-label.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/create-a-new-shipping-label.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

anaharris-ms, 2022-11-17:
Hi @HaraldWang: 

Just a few questions to help address your issue:

1 - Are you unable to publish at all due to this issue?
2 - Is there another issue you are having because you cannot access these two fields via the API?

Any more information on how this is impacting you will be of help!

Thanks!

anaharris-ms, 2023-01-31:
Closing out due to inactivity.  Please reopen if still needed.  Thanks for reaching out!

closed 2023-01-31 by anaharris-ms