MicrosoftDocs/windows-driver-docs #3328: I believe there's an error with the descriptions given for commands,

DermotBohan, 2022-11-14:
[Enter feedback here]
From the page:

Copy
pnputil /export-driver oem6.inf .                                  
Export all driver packages

Copy
pnputil /export-driver * c:\backup                                 
Disables device specified by device instance ID

^^^Explanation appears to have been mixed up.
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 41c58204-8ae9-7822-df54-6e33af08f3c7
* Version Independent ID: 3ded15a1-1b56-e571-adfd-383b27f64217
* Content: [PnPUtil Examples - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-examples?source=recommendations)
* Content Source: [windows-driver-docs-pr/devtest/pnputil-examples.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/pnputil-examples.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

mhopkins-msft, 2022-11-14:
Hi @DermotBohan,

Your timing is amazing! We're actually in the middle of updating this page right now. The description actually applies the example after. Hopefully, the new formatting will make that clearer. It should be fixed and published later today.

Thanks for reaching out to help make the docs better.

 ~ *Mark*

closed 2022-11-14 by mhopkins-msft

hiram1903, 2023-04-05:
> [Enter feedback here] From the page:
>
> Copy pnputil /export-driver oem6.inf . Export all driver packages
>
> Copy pnputil /export-driver * c:\backup Disables device specified by device instance ID
>
> ## [...]