MicrosoftDocs/windows-driver-docs #3329: What should be used instead of AddDevice?

thxkiwi, 2022-11-15:
https://github.com/MicrosoftDocs/windows-driver-docs/issues/3019 was closed with the resolution that DDInstall32 and AddDevice should be used only by In-box drivers and was removed from the driver samples.

I observe that without using any such construct/directives the tableaudiosample never actually creates a device.

If DDInstall32/AddDevice are to be used only by In-box drivers, how does one add a device such a sysvad?

tedhudek, 2022-11-15:
Hi Michael, can you give me more info on your specific scenario? It's complex enough that giving you the best answer really depends on what you are trying to accomplish.

tedhudek, 2022-11-22:
Thanks Michael.  If your ultimate purpose is just to see the sample in action and the sample is designed to work on a virtual device with no real hardware present, then you could create a root enumerated device that the sample installs on.

To do that, you could use `devcon install tabletaudiosample.inf Root\sysvad_TabletAudioSample`.

If you are creating your own driver based off the sample, then presumably you would have real audio hardware to install your driver on and not need a virtual device.

thxkiwi, 2022-11-23:
Ted,

We make use of a VAD to allow an APO to be installed independently from specific audio hardware. I’m confident we’re not the only commercial product doing so. I’m specifically asking about this case and if the DeviceInstall32/AddDevice method is not the correct method then what should be used?

Michael “Kiwi” Ngarimu
Senior Principal Architect
THX Ltd.
main(i){return putchar((0x584854406977694B>>(8*(i-1)))&0xFF)&&main(++i);}

tedhudek, 2022-11-30:
Hi Michael, thanks for your patience.  I have a few things for you to try.

First off, you'll want to follow the guidance in this section -- https://learn.microsoft.com/en-us/windows-hardware/drivers/audio/operating-system-upgrades#hardware-id-device-instance-name.

Second, `pnputil /add-driver` and `devcon dp_add` both act like you right clicked on the inf file and did the install that way, which assumes there is a devnode that needs the driver or that the inf will add the devnode. What you likely want is `devcon install`, which is like going into Device Manager and doing **Add legacy hardware** in that it creates a devnode under `root\00XX` and installs the INF on that devnode.

DOMARS, 2022-12-27:
@thxkiwi - We are working to lock down some information of the use of how the [setupapi.h](https://learn.microsoft.com/windows/win32/api/setupapi/) header can be used in a reasonable way for this.

This is the rough API sequence we are looking at. 
•	Create the root enumerated device as a phantom device
•	Give that device unique hardware and/or compatible IDs
•	Register the root enumerated device to make it usable

Working on some details for GUIDs and such. I will provide any updates on this work when I have them.

DOMARS, 2023-02-01:
@thxkiwi - The best information I have on this, is that the use of the APIs in [setupapi.h header](https://learn.microsoft.com/en-us/windows/win32/api/setupapi/) is what has normally been done here.

But it looks like another option might be to use a [SwDevices](https://learn.microsoft.com/en-us/windows/win32/api/swdevice) instead of a root enumerated devices. The [SwDeviceCreate function](https://learn.microsoft.com/en-us/windows/win32/api/swdevice/nf-swdevice-swdevicecreate) is used to create a software device. But there is an open question on how that would migrate in OS upgrades and interact with existing audio code in Windows. 

I will @ mention you when I have additional information on this newer approach, but that may be a bit, as there may need to be some additional testing, and that testing may take some time.  

As I mentioned to use the setup API approach, I would: 

* Create the root enumerated device as a phantom device
* Give that device unique hardware and/or compatible IDs
* Register the root enumerated device to make it usable

For more information, see:

[SetupDiCreateDeviceInfoList](https://learn.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdicreatedeviceinfolist)
[SetupDiCreateDeviceInfoW](https://learn.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdicreatedeviceinfow)
[SetupDiSetDeviceRegistryProperty](https://learn.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdisetdeviceregistrypropertya)
[SetupDiCallClassInstaller](https://learn.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdicallclassinstaller)
[SetupDiDestroyDeviceInfoList](https://learn.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdidestroydeviceinfolist)

This has been a very useful issue to raise, so that we can better clarify the best approach here and expand the docs to describe this with more clarity.  Thanks for filing this issue and I will pass on any additional information as I have it.

closed 2023-02-01 by DOMARS

thxkiwi, 2023-02-01:
Don,

Thanks for the update. As far as utilizing APIs (Setupapi or SwDevices), since this is all being done as part of an install via INFs what is the recommendation for providing software to make these API calls, while still complying with DCH(U)? Is it simply that a virtual audio device is not compliant with DCHU?

Michael "Kiwi" Ngarimu
Senior Principal Architect
THX Ltd.
main(i){return putchar((0x584854406977694B>>(8*(i-1)))&0xFF)&&main(++i);}