MicrosoftDocs/windows-driver-docs #3353: Not able to submit new driver request

neeshupal, 2022-12-19:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a9288fd9-bb24-631a-b6f8-6b2f5313fd17
* Version Independent ID: a9288fd9-bb24-631a-b6f8-6b2f5313fd17
* Content: [Create a new hardware submission - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-create)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-submission-create.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-submission-create.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

neeshupal, 2022-12-19:
Hi ,

I I click on https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-create link I landing page is coming completely different from what is mentioned on the Microsoft doc

anaharris-ms, 2022-12-19:
Hi @neeshupal ,

Sorry you are having difficulties. Can you please provide the link to the Microsoft doc that contains that link? 

Thank you!
Anastasia

anaharris-ms, 2023-01-31:
Closing out due to inactivity.  Please reopen if still needed.  Thanks for reaching out!

closed 2023-01-31 by anaharris-ms