MicrosoftDocs/windows-driver-docs #3357: I Disable Signature Verification, but Windows still need digital signature for unsigned driver file

ruoxu233, 2022-12-26:
[Enter feedback here]
Chapter (Use the F8 Advanced Boot Option) working in my Windows10, but Chapter (Attach a Kernel Debugger to Disable Signature Verification) does not working in my Windows10.

error is 577,  Windows cannot verify the digital signature for this file.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 3aeb487d-b749-4f6e-9ff5-71be68bf7a09
* Version Independent ID: 44418645-8c6e-7387-ba3e-02f478816961
* Content: [Installing an Unsigned Driver during Development and Test - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/installing-an-unsigned-driver-during-development-and-test)
* Content Source: [windows-driver-docs-pr/install/installing-an-unsigned-driver-during-development-and-test.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/installing-an-unsigned-driver-during-development-and-test.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

DOMARS, 2023-01-04:
@ruoxu233 - In later versions of Windows, just attaching the kernel debugger may not be sufficient, so we will investigate doc updates to this topic. Also in modern times, secure boot, bit locker and device guard / defender and registry policy settings can impact driver loading. If you do have any of those features, you could disable those and retest, to see if that is sufficient to allow an unsigned driver to load with a debugger attached. In addition, a transport connection may need to be configured and active, when the driver loads. The presence of the catalog file and the preparation of the INF for the unsigned driver, can also impact driver loading.

In general, I would test sign drivers, and I would recommend enabling test signing in recent versions of Windows.  This lab walks through a recommended process for test drivers.
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-#section-3-download-and-build-the-kmdf-echo-driver

DOMARS, 2023-01-27:
@ruoxu233 - I was able to test this, and the key thing to note is that the kernel debugger connection must be active as you are installing the driver. In that the debugger could break into the target system durring the install. In addition, the bcdedit settings must be set as described, but they are not sufficient by themselves. As the topic says:

*Attaching an active kernel debugger to a development or test computer disables load-time signature enforcement for kernel-mode drivers.* 

There certainly can be a unique configuration issue that is complicating any testing and installation of the driver. Sometimes a single failed driver install can lead to other issues. The best approach there is to study the driver install log at `%windir%\inf\setupapi.dev.log.`
As we focus on the docs in these git issues, if you have further questions on your setup, I would recommend the driver Q&A forums, such as [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html).  
Good luck with your driver development and debugger work.

closed 2023-01-27 by DOMARS