MicrosoftDocs/windows-driver-docs #3364: Bug 0x3a - SYSTEM_UNWIND_PREVIOUS_USER

RespectIsEverything, 2023-01-06:
Complete lack of documentation concerning bug code 0x3a.
The only thing mentioned is "This bug check appears very infrequently."

I'm troubleshooting a Hyper-V VM that crashes when installing Windows 11 (any build) to a physical drive.
The BSOD occurs immediately after proceeding from the partitioning screen.

- In the same VM with identical configuration, Windows 11 installs just fine to a virtual disk.
- In the same VM with identical configuration, Windows 10 22h2 installs just fine to the physical drive.
- Upgrading the Windows 10 install to Windows 11 via the Upgrade Assistant also causes the same crash.

I cannot get crash details from the VM. The only log I can access is the 'Hyper-V Worker' log with the following entry:

**Event ID 18590**
'VM 1' has encountered a fatal error. The guest operating system reported that it failed with the following error codes:

VmErrorCode0 0x139
VmErrorCode1 0x3a
VmErrorCode2 0xffffbf8e4645efd0
VmErrorCode3 0xffffbf8e4645ef28
VmErrorCode4 0x0
VmErrorMessage
VmPreOSId 0
VmHvOSId 0x1040a0000271b

The lack of information regarding Bug 0x3a is quite unhelpful.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 89eeac17-9e57-fb32-3e38-5a2b227a2b90
* Version Independent ID: 972049ed-64e2-d37b-0fb1-fbe1d2cf0334
* Content: [Bug Check 0x3A SYSTEM_UNWIND_PREVIOUS_USER - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x3a--system-unwind-previous-user)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x3a--system-unwind-previous-user.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x3a--system-unwind-previous-user.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RespectIsEverything, 2023-01-20:
May I inquire as to any updates on this topic?
Further guidance on how to proceed would certainly be appreciated...

DOMARS, 2023-01-28:
@RespectIsEverything - I have reached out to the dev team as this bug check code has not been documented in the past and no parameters are displayed with !analyze. Over time I'm working to remove the "this occurs infrequently" and add a basic description. There are some of these low numbered bug check codes that are no longer used by Windows, but are used by third parties, which complicates things further. 

This category of error in Windows is typically related to the transition or unwind from different memory types, such as from kernel to user. These errors are normally present in the source code and occur, under specific conditions. Access to the faulting source code is normally required to diagnose and repair, what is typically the mismanagement of kernel memory, or IRQL levels that memory is accessed at.

The normal approach would be to attach a kernel debugger and look at the calling stack, and registers.  And disassemble the instructions if no source is available. Without access to the hyper v area, it sounds like you are not able to get a kernel dump file. With that, you could look at the call stack and see if !analyze could identify a faulting module. Looking at the exception record, the exception handling chain, and the fault areas, as well as the registers, and the disassembled code, can sometimes provide clues.

If that is not available, sometimes looking in the event logs for things that happen right before the crash may provide clues, if the entity that is going to crash is able to know that, and add event log entries.  If the crash is repeatable, increasing the log levels for related entities, may provide additional information.

In your scenario one of the differences seems to be between physical and managed drivers, wrapped in the Intel RST Volume Management. Perhaps double checking that for any updates or known issues or seeing as a test, if it can be replaced with something else, such as different storage system. This Dell article describes, needing to install drivers for that system - > https://www.dell.com/support/kbdoc/en-lv/000188116/intel-11th-generation-processors-no-drives-can-be-found-during-windows-10-installation

It normally is a multi-week process to document a bug check code, so you may want to consider other approaches in the interim. 

I will close this issue for now, but will @ mention you if I'm able to provide any additional information.

closed 2023-01-28 by DOMARS

RespectIsEverything, 2023-01-28:
Thanks for the update.

Some of these troubleshooting steps are a bit more advanced than I'm used to, but I'll nonetheless attempt to attach the kernel debugger in order to produce a dump. Fortunately, your documentation, https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/attaching-to-a-virtual-machine--kernel-mode-, is succinct.

As for your suggestion about loading the RST-VMD drivers:
It is true that they must be loaded when installing Windows OS on the host machine. The partitioning screen will not detect any disks until that driver is loaded.

But the VM's partitioning screen detects the attached disks. My understanding is that the Host automatically passes the RST-VMD drivers onto the Guest VM. 

Furthermore, the crash only occurs with Windows 11 when installing the VM OS onto a secondary physical disk. The crash does not occur when installing Windows 11 onto a virtual disk located on the secondary physical disk. Windows 10 does not crash in either scenario.

I'll see what dumps the kernel debugger produces and post back if it might be helpful towards your documentation efforts.
My sincere gratitude for your response.