MicrosoftDocs/windows-driver-docs #3377: SDK link takes me to the French site.

rconiv, 2023-01-20:
[Enter feedback here]
When I click on either link, it takes me to the french download site.

Debugging Tools for Windows (WinDbg)
Get Debugging Tools for Windows (WinDbg) from the SDK: [Windows SDK](https://developer.microsoft.com/windows/downloads/windows-10-sdk). Use the download link on the [Windows SDK](https://developer.microsoft.com/windows/downloads/windows-10-sdk) page, as the Debugging Tools for Windows are not available as part of Visual Studio.

https://developer.microsoft.com/fr-fr/windows/downloads/windows-sdk/  I can change the fr-ft to en-us and it goes to the U.S. site.
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 162fe2fe-9216-f4c5-9f91-3df8776bb99e
* Version Independent ID: 013dbf6b-0790-9c11-24f4-dbf24d64401e
* Content: [Download Debugging Tools for Windows - WinDbg - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-tools)
* Content Source: [windows-driver-docs-pr/debugger/debugger-download-tools.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-tools.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

tedhudek, 2023-01-20:
Thanks for the report @rconiv.  I don't repro this behavior, clicking the links from the doc page.  Could it be related to your browser settings?

rconiv, 2023-01-20:
> CAUTION: This email originated from outside of UniversalLanguage. DO NOT click links or open attachments unless you recognize the sender (e.g. senders's domain) and know the content is safe. Please contact your direct supervisor/manager or UniversalLanguage IT Department, if you are unsure if it is safe to click on the link, or open the attachment.

Wouldn’t think so, since I am in the US, and everything is set to English.  There is no other languages installed.

Best Regards,
Jon Mercer
HelpDesk Support
Universal Language Service, Inc.

***@***.***?anonymous&ep=signature>
Book time to meet with ***@***.***?anonymous&ep=signature>

Phone:  1 (888) 462-0500 • Direct: 1 (425) 691-3200
Email: ***@***.******@***.***>
Address:  929 108th Ave NE, Suite 710, Bellevue, WA 98004
***@***.***<http://www.universallanguageservice.com/>
Over-the-Phone<https://urldefense.proofpoint.com/v2/url?u=http-3A__universallanguageservice.com_services_over-2Dthe-2Dphone-2Dinterpretation_&d=DwMFAg&c=CO1ThuriqckIOSHVQ0MREA&r=iufB2MB82pXfSSOoYvfqR4lfTI2ZivYikDDJBAvUJ7w&m=ru1lSxAsm6NuqURnDCn08tlj1kOuPlyLAY30nz358gk&s=XmmBjcpSo1fnvKuGrwBLzBiF9up4ASWFXgtJ2ocHXjs&e=> | Video Remote<https://urldefense.proofpoint.com/v2/url?u=http-3A__universallanguageservice.com_services_video-2Dinterpretation_&d=DwMFAg&c=CO1ThuriqckIOSHVQ0MREA&r=iufB2MB82pXfSSOoYvfqR4lfTI2ZivYikDDJBAvUJ7w&m=ru1lSxAsm6NuqURnDCn08tlj1kOuPlyLAY30nz358gk&s=aGZGa3f5O7uhDjU1tKJPFHM3I6X_W1JWyE52A1tR79g&e=> | In-Person<https://urldefense.proofpoint.com/v2/url?u=http-3A__universallanguageservice.com_services_in-2Dperson-2Dinterpretation_&d=DwMFAg&c=CO1ThuriqckIOSHVQ0MREA&r=iufB2MB82pXfSSOoYvfqR4lfTI2ZivYikDDJBAvUJ7w&m=ru1lSxAsm6NuqURnDCn08tlj1kOuPlyLAY30nz358gk&s=1TptnKdPQCm-AshEFJN3O1z7TcGMTI84qSInlEomNkc&e=> | Document Translation<https://urldefense.proofpoint.com/v2/url?u=http-3A__universallanguageservice.com_services_written-2Dtranslation_&d=DwMFAg&c=CO1ThuriqckIOSHVQ0MREA&r=iufB2MB82pXfSSOoYvfqR4lfTI2ZivYikDDJBAvUJ7w&m=ru1lSxAsm6NuqURnDCn08tlj1kOuPlyLAY30nz358gk&s=ozqNpwid-4VhHi69dg6wkNTCv_Ol-7UMFuvv-CxSpnA&e=> | 200+ Languages<https://urldefense.proofpoint.com/v2/url?u=http-3A__universallanguageservice.com_customer-2Dsupport_languages_&d=DwMFAg&c=CO1ThuriqckIOSHVQ0MREA&r=iufB2MB82pXfSSOoYvfqR4lfTI2ZivYikDDJBAvUJ7w&m=ru1lSxAsm6NuqURnDCn08tlj1kOuPlyLAY30nz358gk&s=Logy8K314wklX_nS4jxeKFA9K8swoh0SNTcn1P7ePCI&e=>