MicrosoftDocs/windows-driver-docs #3379: Does not form where windbg preview is installed

nopeless, 2023-01-20:
I somehow cannot find any resources related to the installation location of windbg. Right clicking it in the search bar does not show any options to show the location of the shortcut nor the location of the actual executable

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 8ea1adb7-1680-611f-a605-f8f10c7e26d8
* Version Independent ID: 37a27cc0-a7e9-adc0-2ce8-d9eb13df9c21
* Content: [WinDbg Preview - installation - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-install-preview)
* Content Source: [windows-driver-docs-pr/debugger/windbg-install-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/windbg-install-preview.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-01-27:
@nopeless - Store Apps manage their install locations a bit behind the scenes. You should be able to use File Explorer to search the entire drive to see where it ended up on your PC. 

For the file extensions this WinDbg Product Feedback issue > https://github.com/microsoftfeedback/WinDbg-Feedback/issues/32 mentions this directory may be used for debugger extensions - `%LOCALAPPDATA%\Dbg\EngineExtensions`.

The mosty popular feedback item for the debugger is to ship the debugger outside of the store. Doing this would likely allow more control of where it installs.

https://github.com/microsoftfeedback/WinDbg-Feedback/issues/19

You can decide if you want to upvote any of the existing product issues, or create a new one.

Good luck with your Windows Debugging and thanks for the interest in the Windows debugger documentation.

nopeless, 2023-01-27:
My request was that this should be mentioned in the docs @DOMARS