MicrosoftDocs/windows-driver-docs #3380: Unclear how to call USB3HWVerifierAnalyzer.exe with filtering events with VID/PID

sdbbs, 2023-01-22:
I am trying to consult https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/how-to-retrieve-information-about-a-usb-device ( https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/how-to-retrieve-information-about-a-usb-device.md ) about `USB3HWVerifierAnalyzer.exe`. It mentions:

> -v <VendorID> | Logs all hardware verifier events for the specified VendorID.  
> -p <ProductID> | Logs all hardware verifier events for the specified ProductID.  
>
> If you want to filter events for a particular hardware, specify the VendorId and ProductId of the hardware.

I simply cannot get this to work. I have called the tool like this in the Administrative command prompt:

    C:\>"C:\Program Files (x86)\USBTest\x86\USB3HWVerifierAnalyzer.exe"

... and the only way I can obtain any readable output is with:

    C:\>"C:\Program Files (x86)\USBTest\x86\USB3HWVerifierAnalyzer.exe" -f AllEvents.etl /v > AllEventsOutput.txt

... and in this case I get ALL possible events (with all kinds of VID/PID) - which is expected, as I have not specified anything in the call to `USB3HWVerifierAnalyzer.exe`. 

If I leave the `/v` switch out (i.e. I use only `... -f AllEvents.etl > AllEventsOutput.txt`), then I simply get this:

```none

Session Name : TraceSessionSunJan220336482023

**********************************************************************************************
Test Results:

Below is a report of all the Hardware verifier events encountered. The Key field refers to a
Controller, hub, or device. During the lifetime of this utility, all unfiltered events are
captured in file AllEvents.etl (in current directory). The Key in below report can be used to
filter events in AllEvents.etl file

**********************************************************************************************

```

... and nothing else.

Ok, now - if I try to filter by VID/PID, either with:

    C:\>"C:\Program Files (x86)\USBTest\x86\USB3HWVerifierAnalyzer.exe" -v 428A -p 000A

or with:

    C:\>"C:\Program Files (x86)\USBTest\x86\USB3HWVerifierAnalyzer.exe" -v 0x428A  -p 0x000A

... where 0x428A and 0x000A are the test VID and PID on my device, I get absolutely NO filtering - ALL USB events, regardless of VID/PID, are visible in AllEventsOutput.txt (with `...  /v > AllEventsOutput.txt` -- without it, as mentioned, I get an empty log with just a heading text).

Can you please state in the documentation the *exact* command line, to obtain filtering by VID/PID - and the expected output? To be fair, you do have an output given on that page, that is:

> Here is an example output from the hardware verifier tool:
> 
> ```
> Session Name : TraceSessionWedJun062021182012
> 
> Attempting to start session TraceSessionWedJun062021182012...
> Trace Session created...Status : 0
> 
> Provider Enable Success, Status : 0
> ...
> ```

.... however you have not given the actual command that was ran to obtain that output, and I cannot tell if that is supposed to show filtered output or not.

Also, there is this:

> Open AllEvents.etl in Netmon and select Add <field_name> to display filter to filter events by controller, hub, and device.

I did open AllEvents.etl in Microsoft Network Monitor 3.4 (what I get when I type `netmon` in Windows 10) - and I get this:

[image]

I see no "Add" button here - and furthermore, the log seems useless for USB, as all I can see are "Net Event" and "Net Event" - not even VID/PID information.

Can you please add a screenshot of Netmon in that page, and circle in red where exactly people are supposed to click, to obtain some data relevant to USB, like VID&PID?

mhopkins-msft, 2023-01-25:
Hi @sdbbs 

I am checking with our engineering team regarding your questions. Meanwhile, you may want to post your question in the following forums, where other driver developers and Microsoft engineers can be found.

- [Windows development Q&A](https://learn.microsoft.com/answers/products/windows)
- [Windows Hardware WDK and driver development forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk)
- [OSR NTDEV forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV)

I will post more information here when I get it from the engineering team.

Thanks,
 ~ *Mark*

mhopkins-msft, 2023-02-02:
Hi @sdbbs

Here is the response I got from the engineering team:
__________

When filtering for VID/PID it will not output all events for that VID/PID, but only hardware verifier events, which are typically errors like invalid descriptors, error recovery events, and so on.

The AllEvents ETL will always contain all ETW events for all devices.  It is not affected by the -v/-p switches.

Also, to view the ETW events in netmon you need install the USB ETW parsers:

    https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/how-to-install-netmon-and-the-netmon-usb-parser    

Here's output from my testing of usb3hwverifieranalyzer.  Note that while I'm filtering for a USB mass storage device, the event is logged against its parent hub, which is what is displayed in the filtered output:

```console
usb3hwverifieranalyzer.exe -v 0781 -p 5595

Session Name : TraceSessionFriJan271351112023

Attempting to start session TraceSessionFriJan271351112023...
Trace Session created...Status : 0

Provider Enable Success, Status : 0

Provider Enable Success, Status : 0

Provider Enable Success, Status : 0

Provider Enable Success, Status : 0

Provider Enable Success, Status : 0

Provider Enable Success, Status : 0
13319329877.425596: (UsbHub3/179)
        Event Message: Client Initiated Recovery Action
        VendorID/ProductID: 0x5e3/0x612
        DeviceInterfacePath: \??\USB#VID_05E3&PID_0612#6&130491ac&0&4#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
        DeviceDescription: Generic SuperSpeed USB Hub
        PortPath:  0x12, 0x4, 0x0, 0x0, 0x0, 0x0

Provider disable Success, Status : 0

Provider disable Success, Status : 0

Provider disable Success, Status : 0

Provider disable Success, Status : 0

Provider disable Success, Status : 0

Provider disable Success, Status : 0

Session Stopped...Status : 0
```
__________

I've updated the topic to include this information.

Thanks,
 ~ *Mark*

closed 2023-02-02 by mhopkins-msft