MicrosoftDocs/windows-driver-docs #3381: Client-side rendering with a driver mismatch

printric, 2023-01-26:
If the suggested regkey HKLM\SYSTEM\CurrentControlSet\Control\Print\PrinterName\PrinterDriverData\DriverPolicy is applied to queue on Windows Server 2016 printserver, mapped on a client running Windows 10 21H2 >October Patch (maybe even earlier),  the queue will not be usable after a restart of the client spooler service or a reboot of this machine.

Is this regkey still a valid workaround for a driver mismatch?

Document: Known Issues with Client-Side Rendering
Path: windows-driver-docs-pr/print/known-issues-with-client-side-rendering.md

tedhudek, 2023-01-26:
Hi @printric, could you please provide a link to the docs page you are referencing?  Thanks!

printric, 2023-01-26:
https://learn.microsoft.com/en-us/windows-hardware/drivers/print/known-issues-with-client-side-rendering
https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/known-issues-with-client-side-rendering.md

barrygolden, 2023-02-01:
@printric, thanks for reaching out to the Windows Driver Developer Documentation team. You may have success posting your question about client-side rendering with a driver mismatch in the following forums, where other driver developers and Microsoft engineers can be found. 

[Microsoft Q&A: Windows Hardware development](https://docs.microsoft.com/answers/topics/windows-hardware.html)

[OSR Developer Community](https://community.osr.com/)

closed 2023-02-01 by barrygolden