MicrosoftDocs/windows-driver-docs #3383: command line options do not work

chupperdg, 2023-01-27:
WinDbgX /y srv*c:\sym*http://msdl.microsoft.com/download/symbols

From:
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-command-line-preview

/y -y and many others are broken compared to orig windbg.
`:000> .sympath 
Symbol search path is: srv*
Expanded Symbol search path is: cache*;SRV*https://msdl.microsoft.com/download/symbols

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
`
We shouldn't have to change all these options via UI.

DOMARS, 2023-01-27:
@chupperdg - There are differences between the command line load options between the classic and the new debugger. 

If you think there are specific command line options that should behave as they did in the classic debugger, you can create a git issue on the debugger's team github site describing those differences.

http://aka.ms/dexex 

There are a few already there related to startup - for example locating extensions, and launching UWP apps.
https://github.com/microsoftfeedback/WinDbg-Feedback/issues/32
https://github.com/microsoftfeedback/WinDbg-Feedback/issues/36

Some items such as symbol path can also be set via an environment value before the debugger is loaded. This is described in: https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/symbol-path

There is also the concept of workspaces in the new debugger that allows you to configure many things, as sets of configuration information. That is described here: https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-setup-preview

Thanks for your interest in the debugger docs and the new debugger.

closed 2023-01-27 by DOMARS

chupperdg, 2023-01-27:
the specific one I'm complaining about is /y.
the doc link says it should work for windbgx 
but it does not.
I've been trying this since the debugger came out, it has never worked for me.

The workspace doesn't seem to provide a way to set the symbols in a workspace file and then pass that file to windbgx on the command line,  Saving workspace produces a file that does not contain what is specified in the symbols UI