MicrosoftDocs/windows-driver-docs #3388: Fix values on "Keyboard Identifiers and Input Method Editors for Windows" page

DJm00n, 2023-01-30:
Some values on page [Keyboard Identifiers and Input Method Editors for Windows](https://learn.microsoft.com/en-us/windows-hardware/manufacture/desktop/windows-language-pack-default-values) looks wrong:
Bulgarian `0x0030402` -> `0x0030402`
India `0x000004009` -> `0x00004009`

Also list looks a bit outdated. This is what I got with simple tool on my Windows 10 21H2 19044.2486 with [this simple tool](https://gist.github.com/DJm00n/fa6d71a188289c200b5f92d83b5cf0a3):

| Keyboard                                     | Keyboard identifier (hexadecimal) |
|-----------------------------------------|-------------------------------|
| ADLaM                                        | 0x00140c00                        |
| Albanian                                     | 0x0000041c                        |
| Arabic (101)                                 | 0x00000401                        |
| Arabic (102)                                 | 0x00010401                        |
| Arabic (102) AZERTY                          | 0x00020401                        |
| Armenian Eastern (Legacy)                    | 0x0000042b                        |
| Armenian Phonetic                            | 0x0002042b                        |
| Armenian Typewriter                          | 0x0003042b                        |
| Armenian Western (Legacy)                    | 0x0001042b                        |
| Assamese - INSCRIPT                          | 0x0000044d                        |
| Azerbaijani (Standard)                       | 0x0001042c                        |
| Azerbaijani Cyrillic                         | 0x0000082c                        |
| Azerbaijani Latin                            | 0x0000042c                        |
| Bangla                                       | 0x00000445                        |
| Bangla - INSCRIPT                            | 0x00020445                        |
| Bangla - INSCRIPT (Legacy)                   | 0x00010445                        |
| Bashkir                                      | 0x0000046d                        |
| Belarusian                                   | 0x00000423                        |
| Belgian (Comma)                              | 0x0001080c                        |
| Belgian (Period)                             | 0x00000813                        |
| Belgian French                               | 0x0000080c                        |
| Bosnian (Cyrillic)                           | 0x0000201a                        |
| Buginese                                     | 0x000b0c00                        |
| Bulgarian                                    | 0x00030402                        |
| Bulgarian (Latin)                            | 0x00010402                        |
| Bulgarian (Phonetic Traditional)             | 0x00040402                        |
| Bulgarian (Phonetic)                         | 0x00020402                        |
| Bulgarian (Typewriter)                       | 0x00000402                        |
| Canadian French                              | 0x00001009                        |
| Canadian French (Legacy)                     | 0x00000c0c                        |
| Canadian Multilingual Standard               | 0x00011009                        |
| Central Atlas Tamazight                      | 0x0000085f                        |
| Central Kurdish                              | 0x00000492                        |
| Cherokee Nation                              | 0x0000045c                        |
| Cherokee Phonetic                            | 0x0001045c                        |
| Chinese (Simplified) - US                    | 0x00000804                        |
| Chinese (Simplified, Singapore) - US         | 0x00001004                        |
| Chinese (Traditional) - US                   | 0x00000404                        |
| Chinese (Traditional, Hong Kong S.A.R.) - US | 0x00000c04                        |
| Chinese (Traditional, Macao S.A.R.) - US     | 0x00001404                        |
| Czech                                        | 0x00000405                        |
| Czech (QWERTY)                               | 0x00010405                        |
| Czech Programmers                            | 0x00020405                        |
| Danish                                       | 0x00000406                        |
| Devanagari - INSCRIPT                        | 0x00000439                        |
| Divehi Phonetic                              | 0x00000465                        |
| Divehi Typewriter                            | 0x00010465                        |
| Dutch                                        | 0x00000413                        |
| Dzongkha                                     | 0x00000C51                        |
| English (India)                              | 0x00004009                        |
| Estonian                                     | 0x00000425                        |
| Faeroese                                     | 0x00000438                        |
| Finnish                                      | 0x0000040b                        |
| Finnish with Sami                            | 0x0001083b                        |
| French                                       | 0x0000040c                        |
| Futhark                                      | 0x00120c00                        |
| Georgian (Ergonomic)                         | 0x00020437                        |
| Georgian (Legacy)                            | 0x00000437                        |
| Georgian (MES)                               | 0x00030437                        |
| Georgian (Old Alphabets)                     | 0x00040437                        |
| Georgian (QWERTY)                            | 0x00010437                        |
| German                                       | 0x00000407                        |
[... 122 lines cut ...]
| Tifinagh (Basic)                             | 0x0000105f                        |
| Tifinagh (Extended)                          | 0x0001105f                        |
| Traditional Mongolian (Standard)             | 0x00010850                        |
| Turkish F                                    | 0x0001041f                        |
| Turkish Q                                    | 0x0000041f                        |
| Turkmen                                      | 0x00000442                        |
| US                                           | 0x00000409                        |
| US English Table for IBM Arabic 238_L        | 0x00050409                        |
| Ukrainian                                    | 0x00000422                        |
| Ukrainian (Enhanced)                         | 0x00020422                        |
| United Kingdom                               | 0x00000809                        |
| United Kingdom Extended                      | 0x00000452                        |
| United States-Dvorak                         | 0x00010409                        |
| United States-Dvorak for left hand           | 0x00030409                        |
| United States-Dvorak for right hand          | 0x00040409                        |
| United States-International                  | 0x00020409                        |
| Urdu                                         | 0x00000420                        |
| Uyghur                                       | 0x00010480                        |
| Uyghur (Legacy)                              | 0x00000480                        |
| Uzbek Cyrillic                               | 0x00000843                        |
| Vietnamese                                   | 0x0000042a                        |
| Wolof                                        | 0x00000488                        |
| Yoruba                                       | 0x0000046a                        |

See [diff with current version](https://www.diffchecker.com/n0ZkL2ZL/).

mhopkins-msft, 2023-01-31:
Hi Dimitriy,

Thank you for reporting this, and for the handy tool you wrote! I was able to build and run the tool to verify your results.

FYI, instances of `layoutName` should be changed to `layoutDisplayName` in the `GetKeyboardLayoutDisplayName` function.

I'll keep this tool around for future use. I really appreciate your continued help improving the docs.

The doc page has been updated and should be published later today.

Thanks again!
 ~ *Mark*

closed 2023-01-31 by mhopkins-msft

DJm00n, 2023-01-31:
>FYI, instances of layoutName should be changed to layoutDisplayName in the GetKeyboardLayoutDisplayName function.
oh, my bad. fixed it on gist.

>The doc page has been updated and should be published later today.
Thank you for a immediate feedback!