MicrosoftDocs/windows-driver-docs #3393: Fix TSF IME values on "Keyboard Identifiers and Input Method Editors for Windows" page.

DJm00n, 2023-02-01:
I [wrote another tool ](https://gist.github.com/DJm00n/d9706fe331704db656d32a233fc7e540)to generate TSF IMEs for [Keyboard Identifiers and Input Method Editors for Windows](https://learn.microsoft.com/en-us/windows-hardware/manufacture/desktop/windows-language-pack-default-values) page.

See #3388 for my previous tool.

| Language/Region | Input profile (Language and keyboard pair) |
|-|-|
| Amharic - Ethiopia | am-ET: Amharic Input Method 2 045e:{7C472071-36A7-4709-88CC-859513E583A9}{9A4E8FC7-76BF-4A63-980D-FADDADF7E987} |
| Bengali - India | bn-IN: Bangla Phonetic 0445:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{C7F2F349-36A2-4F2E-977B-5C18BAF1AE7D} |
| Chinese (Simplified) - China | zh-CN: Microsoft Wubi 0804:{6A498709-E00B-4C45-A018-8F9E4081AE40}{82590C13-F4DD-44F4-BA1D-8667246FDF8E} |
| Chinese (Simplified) - China | zh-CN: Microsoft Pinyin 0804:{81D4E9C9-1D3B-41BC-9E6C-4B40BF79E35E}{FA550B04-5AD7-411F-A5AC-CA038EC515D7} |
| Chinese (Traditional) - Taiwan | zh-TW: Microsoft Changjie 0404:{531FDEBF-9B4C-4A43-A2AA-960E8FCDC732}{4BDF9F03-C7D3-11D4-B2AB-0080C882687E} |
| Chinese (Traditional) - Taiwan | zh-TW: Microsoft Quick 0404:{531FDEBF-9B4C-4A43-A2AA-960E8FCDC732}{6024B45F-5C54-11D4-B921-0080C882687E} |
| Chinese (Traditional) - Taiwan | zh-TW: Microsoft Bopomofo 0404:{B115690A-EA02-48D5-A231-E3578D2FDF80}{B2F9C502-1742-11D4-9790-0080C882687E} |
| Chinese (Traditional) - Taiwan | zh-TW: Chinese Traditional DaYi 0404:{E429B25A-E5D3-4D1F-9BE3-0C608477E3A1}{037B2C25-480C-4D7F-B027-D6CA6B69788A} |
| Chinese (Traditional) - Taiwan | zh-TW: Chinese Traditional Array 0404:{E429B25A-E5D3-4D1F-9BE3-0C608477E3A1}{D38EFF65-AA46-4FD5-91A7-67845FB02F5B} |
| Gujarati - India | gu-IN: Gujarati Phonetic 0447:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{CDB41983-C911-4685-98CC-FC720BBF8FC8} |
| Hindi - India | hi-IN: Hindi Phonetic 0439:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{26FA8EBD-97DB-4F65-BF35-5E219EBD0079} |
| Japanese - Japan | ja-JP: Microsoft IME 0411:{03B5835F-F03C-411B-9CE2-AA23E1171E36}{A76C93D9-5523-4E90-AAFA-4DB112F9AC76} |
| Kannada - India | kn-IN: Kannada Phonetic 044b:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{0F420707-90F7-4E1B-B9F2-91FF3CA9D990} |
| Korean - Korea | ko-KR: Microsoft IME 0412:{A028AE76-01B1-46C2-99C4-ACD9858AE02F}{B5FE1F02-D5F2-4445-9C03-C568F23C99A1} |
| Korean - Korea | ko-KR: Microsoft Old Hangul IME 0412:{A1E2B86B-924A-4D43-80F6-8A820DF7190F}{B60AF051-257A-46BC-B9D3-84DAD819BAFB} |
| Malayalam - India | ml-IN: Malayalam Phonetic 044c:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{6346F0DC-2FE4-40E6-99D5-7362723C3780} |
| Marathi - India | mr-IN: Marathi Phonetic 044e:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{DBEB227A-4EF6-45A6-85B5-B1A03D95BB66} |
| Nepali - Nepal | ne-NP: Hindi Phonetic 0461:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{7912A487-286C-4825-A608-F676C630B733} |
| Odia - India | or-IN: Odia Phonetic 0448:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{607EBD43-EA66-4CAC-A0B1-40F1D147364B} |
| Punjabi - India | pa-IN: Punjabi Phonetic 0446:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{59139512-9763-4E5D-B8E5-0AC1C3BAE1AE} |
| Tamil - India | ta-IN: Tamil Phonetic 0449:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{045CBA61-4228-4F62-AC0C-F725470D7BFC} |
| Telugu - India | te-IN: Telugu Phonetic 044a:{81EA0A17-AA39-455B-BA20-EA79A8F98966}{9F0CDC2B-12B2-4776-A144-F79901A9F8DA} |
| Tigrinya - Ethiopia | ti-ET: Tigrinya Input Method 0473:{E429B25A-E5D3-4D1F-9BE3-0C608477E3A1}{3CAB88B7-CC3E-46A6-9765-B772AD7761FF} |
| Vietnamese - Vietnam | vi-VN: Vietnamese Number Key-Based 042a:{C2CB2CF0-AF47-413E-9780-8BC3A3C16068}{591AE943-56BE-48F6-8966-06B43915CC5A} |
| Vietnamese - Vietnam | vi-VN: Vietnamese Telex 042a:{C2CB2CF0-AF47-413E-9780-8BC3A3C16068}{5FB02EC5-0A77-4684-B4FA-DEF8A2195628} |
| Yi - China | ii-CN: Yi Input Method 0478:{E429B25A-E5D3-4D1F-9BE3-0C608477E3A1}{409C8376-007B-4357-AE8E-26316EE3FB0D} |

DJm00n, 2023-02-01:
@mhopkins-msft please take a look ))

mhopkins-msft, 2023-02-01:
@DJm00n 

I've incorporated your updates into the topic and grabbed myself a copy of your new tool. Thanks again, Dimitriy! 

 - *Mark*

closed 2023-02-01 by mhopkins-msft