MicrosoftDocs/windows-driver-docs #3396: Fix locale and keyboard values on "Input locales" page

DJm00n, 2023-02-02:
I have finished and [put the tool together on github](https://github.com/DJm00n/keyboardlayoutlist) - now it can generate these three tables:
- [Keyboard identifiers](https://learn.microsoft.com/windows-hardware/manufacture/desktop/windows-language-pack-default-values#keyboard-identifiers)
- [Input method editors](https://learn.microsoft.com/windows-hardware/manufacture/desktop/windows-language-pack-default-values#input-method-editors)
- [Input locales](https://learn.microsoft.com/windows-hardware/manufacture/desktop/default-input-locales-for-windows-language-packs#input-locales)

[Input locales](https://learn.microsoft.com/windows-hardware/manufacture/desktop/default-input-locales-for-windows-language-packs#input-locales) table should be updated on learn.microsoft.com.

[Input method editors](https://learn.microsoft.com/windows-hardware/manufacture/desktop/windows-language-pack-default-values#input-method-editors) have formatting updates - added parentheses for input profile string.

PS: This was fun! 😃 

Related issues #3393 #3388.

@mhopkins-msft

DJm00n, 2023-02-02:
Please also update this text accordingly:
>The input profiles are made up of a [language identifier](https://go.microsoft.com/fwlink/?LinkId=63026) and a [keyboard identifier](https://learn.microsoft.com/en-us/windows-hardware/manufacture/desktop/windows-language-pack-default-values?view=windows-11). For example, the Arabic (Algerian) input profile is 1401:00020401, where 1401 is the hexadecimal identifier of the language: Arabic (Algeria) and 00020401 is the hexadecimal identifier of the keyboard: Arabic 101.

>When the user first identifies the time and date format (User Locale) as Algeria, Windows sets up both the primary input profile, and a secondary input profile: French (France) with French keyboard. The secondary input profile can help the user by providing a keyboard with a Latin character set for tasks that require it, such as filling out email addresses. Some character sets (like CHS IME) have a Latin character set built in.

mhopkins-msft, 2023-02-03:
Table updated, thanks!

What did you have in mind for the text updates?

 ~ *Mark*

DJm00n, 2023-02-03:
>What did you have in mind for the text updates?

It should be:
```
The input profiles are made up of a [language identifier](https://learn.microsoft.com/en-us/openspecs/windows_protocols/ms-lcid/70feba9f-294e-491e-b6eb-56532684c37f) and a [keyboard identifier](https://learn.microsoft.com/en-us/windows-hardware/manufacture/desktop/windows-language-pack-default-values). For example, the *Arabic (Algeria)* input profile is *1401:00020401*, where *1401* is the hexadecimal identifier of the language *Arabic (Algeria)* (language tag *ar-DZ*) and *00020401* is the hexadecimal identifier of the *Arabic (102) AZERTY keyboard*.

When the user first identifies the time and date format (User Locale) as Algeria, Windows sets up both the primary input profile, and a secondary input profile: *French (France)* with *French keyboard*. The secondary input profile can help the user by providing a keyboard with a Latin character set for tasks that require it, such as filling out email addresses. Some input methods (like Microsoft IME) have a Latin character set built in.
```

DJm00n, 2023-02-03:
I did [update](https://github.com/DJm00n/keyboardlayoutlist/commit/46c9a49a15c561f6b434862c4a18023e88aa5c1e) to locale name format - now it will better fit to what we see in Windows:

[image]

mhopkins-msft, 2023-02-03:
There are a couple of updates I'd like to make to your tool but I don't have permission to create branches on your repo.

DJm00n, 2023-02-03:
I just sent you an invitation

mhopkins-msft, 2023-02-03:
Everything is updated, including a PR for the tool.

Thanks, Dimitriy!

DJm00n, 2023-02-08:
Hi @mhopkins-msft!
Sorry to bother you, but I'm still seeing the old "Language - Region" locale name format in the table at https://learn.microsoft.com/. Is this normal?

mhopkins-msft, 2023-02-08:
Hi @DJm00n,

Thank you for following up. I pushed my changes to the repo but didn't finish the final step. It should be published today. Please let me know if you don't see the changes in the next 24 hours.

Thanks!
 ~ *Mark*

DJm00n, 2023-02-09:
This looks wrong:
[image]

mhopkins-msft, 2023-02-13:
It sure does! There was a weird character in the markdown. I fixed it:

[image]

DJm00n, 2023-06-16:
Hi @mhopkins-msft. I have updated my tool at https://github.com/DJm00n/keyboardlayoutlist and added links to corresponding keyboard layouts. Can you please update corresponding pages on https://learn.microsoft.com/ ?

DJm00n, 2023-06-16:
@mhopkins-msft I’ve put output in Readme in repo.

mhopkins-msft, 2023-06-16:
Well, we don't want to publish broken links. Can we make it so the app doesn't crash if there is no valid page to link to? Maybe just output the text and not the link?

mhopkins-msft, 2023-06-16:
Oh, it looks like you just don't include that link, ok. I'll just copy the tables from the README.

mhopkins-msft, 2023-06-16:
Pages are updated. Should republish overnight or by Monday.

closed 2023-06-16 by mhopkins-msft