MicrosoftDocs/windows-driver-docs #3398: Docs not up-to-date

GabeNI, 2023-02-03:
[Enter feedback here]

The docs are missing a number of errors that are reported by newer infverif /k, including:

ERROR(1258)  File path "%12%" in registry may be incorrect in some contexts, use %%SystemRoot%% instead.
ERROR(2084) in file.inf, line 42: Replaced legacy pattern in section 'UMPassService_Install' using 'umpass_include_needs'.

I'm sure there are others that I haven't collected yet.

Also says that 2xxx are only warnings, but this is reported as an error with /k.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 48275f87-4086-1ca5-a190-c741f29ad348
* Version Independent ID: 5481885d-3063-527b-a85e-6a898dbcb990
* Content: [INF Validation Errors and Warnings - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings)
* Content Source: [windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

zlockard, 2023-02-08:
Thanks for the feedback.  I'm working on a new iteration of the docs that has the up to date codes and descriptions for each, however it's been delayed a little bit.  I will update this page with some clarifications immediately though.

GabeNI, 2023-02-08:
@zlockard Good to hear.  I only included a couple of examples in here, but I know we've run into even more than these recently when uploading drivers to the hardware dev dashboard.

tedhudek, 2023-02-15:
Thanks for the feedback @GabeNI.  We just added docs for 2084, see PR #3408.  We'll add more in months to come.  Thanks again for your interest in the docs.

closed 2023-02-15 by tedhudek

GabeNI, 2023-02-17:
> Thanks for the feedback @GabeNI. We just added docs for 2084, see PR #3408. We'll add more in months to come. Thanks again for your interest in the docs.

Thanks, Ted!  I appreciate your responsiveness.  Having all of these errors documented will be very helpful.

GabeNI, 2023-09-22:
@tedhudek Sorry to necropost (if this is bad form, lmk and I'll start a new issue).  I still don't see any documentation for 

   `ERROR(1258) File path "%12%" in registry may be incorrect in some contexts, use %%SystemRoot%% instead`

This has come up again internally.  We are interested to know in which contexts this would be incorrect.

zlockard, 2023-09-27:
It will be incorrect if the driver is added to the machine offline, or during any OS upgrade. Basically, %12% gets resolved at the moment you add the driver to the system, so if you're adding it to a mounted OS image and using DISM it may be a path like E:\mountdir\windows\system32\drivers when you're expecting it to be correct when the OS gets booted. %systemroot% is always resolved by whoever is reading the key, so it will be correct in whatever context they're running in

GabeNI, 2023-09-27:
Thanks for the clear explanation.  That makes a lot of sense.

> %systemroot% is always resolved by whoever is reading the key, so it will be correct in whatever context they're running in

Unless you are our buggy driver that is expecting an absolute DOS path.   😆  (Now fixed.)