MicrosoftDocs/windows-driver-docs #3403: Encountered error 1307 when attestation signing

kcbina, 2023-02-08:
Is there a new way to use ClassInstaller?
Because the old method to use the INF ClassInstall32 section now will be rejected by validation process after the end of january.
Following is the error message:

Error 1307 in xxx.inf, line yyy : Found legacy AddReg operation defining class installers (Installer32).
Failed INF validation. INF did not pass Desktop validation (InfVerif /k).
-----------------------------------------
[ClassInstall32]
AddReg = TESTClassAddReg

[TESTClassAddReg]
HKR,,,,"TEST Boards"
HKR,,Installer32,,"NuDAQClassInstaller.dll,ClassInstaller"
HKR,,Icon,,"110"
-----------------------------------------

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 250f020c-5ea1-f053-7c28-119422e92ea2
* Version Independent ID: 3be4e8c2-4fd9-6deb-0ba4-fa76f09b1021
* Content: [INF CopyFiles directive - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/inf-copyfiles-directive)
* Content Source: [windows-driver-docs-pr/install/inf-copyfiles-directive.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-copyfiles-directive.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-02-08:
Hi @kcbina,

According to the [Windows 11, 22H2 policies](https://go.microsoft.com/fwlink/?linkid=2196181), class installers are no longer supported. As the error message says, the driver needs to pasS InfVerif /k.

The article [Porting an INF to follow driver package isolation](https://learn.microsoft.com/en-us/windows-hardware/drivers/develop/porting-inf-to-windows-driver) has some info on one aspect of coinstallers (not exactly the same as class installer, but similar).  What is it that you are trying to accomplish with a class installer?

kcbina, 2023-02-14:
Hi Tedhudek,

Thanks for your reply. 

The class installer is very important to us, because it seems that our driver does not belong to any class that exists in windows now.

https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-reqs#windows-10-attestation-signed-drivers
As the article mentioned, "For Windows 10 Desktop and later systems, you can submit your drivers for attestation signing, which doesn't require HLK testing. However, attestation signing will only work on Windows 10 Desktop and later versions of Windows"

I'm still confused about being rejected on this part since we belong to "desktop driver".
As the desktop driver, is the new rule no longer support class installers either?

Could you tell me that is there any way to configure the dashboard or modfiy inf file to that server knows that the driver is belong to desktop driver?

Many thanks.

Best Regards,
Luke Chen

tedhudek, 2023-02-16:
Hi Luke. What operating systems are you targeting with your driver?

kcbina, 2023-02-16:
Hi Tedhudek,

Please see the reference pic
[image]

tedhudek, 2023-02-16:
Thanks Luke.  So I'll restate my question above -- what specifically are you trying to accomplish with a class installer?

kcbina, 2023-02-17:
Hi Tedhudek,

Thanks again for your support.
The following is the list for our information.

1. Our product is the "Data Acqusition Card" based on PCI/PCIe bus
2. Can't find any existing class suitable for our product so we build our own class in the past
3. We thought we should able to continue using the classinstaller according to the microsoft documentation due to our driver is belong to "desktop driver" not universal driver.
5. Our submit is rejected from MS Partner Center since the end of January
6. Our question is the same as this page but the author's method does not apply to us for all situations
    https://learn.microsoft.com/en-us/answers/questions/1165452/driver-attestation-signing-fails-due-to-installer3

tedhudek, 2023-02-27:
Hi Luke,

Regarding #3, this is incorrect. See my first message with the link to the 22H2 policy.

Not every custom class needs a class installer. If you are not doing anything meaningful in your class installer, maybe you can just drop it and things will be fine with your custom class.

Without knowing what you are trying to accomplish with the class installer, however, I can't help further.

In any case, we are a documentation team and I'm not seeing a documentation problem here, so I am going to close out this issue.  You could always try the forums if you would like further assistance:

* [windows-hardware-wdk](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk.html)
* [OSR's NTDEV forum](https://community.osr.com/categories/ntdev)

Best of luck!

closed 2023-02-27 by tedhudek