MicrosoftDocs/windows-driver-docs #3404: Linked media files don't seem to be working

aidanplunkitt, 2023-02-08:
The two .wvx downloads for "Advanced Driver Debugging" in [this list](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-resources.md#debugger-how-tos) do not play on Windows Media Player Legacy, they seem to hang with a "Connecting to media..." message so I'm assuming this .wvx file just links to a stream somewhere. After spending a minute looking up how to install a modern version of Windows Media Player and realizing [those docs](https://support.microsoft.com/en-us/windows/get-windows-media-player-81718e0d-cfce-25b1-aee3-94596b658287) are out of date (I am running Win11 22H2), I gave up. Is there a reason we can't just provide direct streaming links or .mp4s downloads?

tedhudek, 2023-02-08:
Thanks for the report Aidan.  We'll investigate.

Feedback pertains to https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-resources#debugger-how-tos.

DOMARS, 2023-02-09:
@aidanplunkitt - Thanks for filing this issue pointing out the broken links. The WMX links you mentioned, were to videos from the 2005 WinHec conference that are no longer available. I replaced those to links with links to the Defrag Tools show, which has a number of great episodes on debugging. I also updated the links to the other topics, so that they go to the current recommended resources for debugging. Thanks for your interest in the Windows debugger and helping to improve the debugging documentation.

closed 2023-02-09 by DOMARS