MicrosoftDocs/windows-driver-docs #3407: Request for clarification of information related to forced OS crash related to GPU

pacampbe, 2023-02-09:
When the section on this page says
> "If the OS detects that six (6) or more GPU hangs and subsequent recoveries occur within one (1) minute, the OS bug-checks the computer on the next GPU hang."

Does it mean both [VIDEO_ENGINE_TIMEOUT_DETECTED(0x141)](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x141---video-engine-timeout-detected) and [VIDEO_TDR_TIMEOUT_DETECTED(0x117)](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x117---video-tdr-timeout-detected)?

Also on the page [Testing and debugging TDR during driver development](https://learn.microsoft.com/en-us/windows-hardware/drivers/display/tdr-registry-keys). The debug regkeys mention that the default values 5 TDRs (bc 0x117) in 1 minute will cause a crash. 

> TdrLimitCount
> Specifies the default number of **TDRs (0x117)** that are allowed during the time specified by the TdrLimitTime key without crashing the computer.
> ValueData : Number of TDRs before crashing. The default value is 5.

Are these two pages describing the same events? Can the details be clarified?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 43cf5418-0c36-a7b9-2f4c-d95c69e9a7ca
* Version Independent ID: 7e890bd6-0ff5-a623-e696-1403fee769c5
* Content: [Timeout detection and recovery (TDR) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/display/timeout-detection-and-recovery)
* Content Source: [windows-driver-docs-pr/display/timeout-detection-and-recovery.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/timeout-detection-and-recovery.md)
* Product: **windows-hardware**
* Technology: **display**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2023-02-23:
Thanks for the great questions, @pacampbe. I've updated the page with clarifications/corrections and added links between the two pages you cite to tie the info together.

closed 2023-02-23 by lorihollasch

pacampbe, 2023-02-23:
Thank you for updating the documentation, it helps.