MicrosoftDocs/windows-driver-docs #3417: More detail for running/finding this tool

robunoki-microsoft, 2023-02-21:
It would be helpful if more context was provided on running this command (and other WDK commands). The command is not on the path unless you are running it from the Visual Studio developer command prompt.  Additionally, why not just list the path to the command?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a022532b-d70d-ac23-d31a-a13313714465
* Version Independent ID: bd640550-ec35-0656-1e5a-8264e84d31fb
* Content: [ComputerHardwareIds Overview - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/computerhardwareids-overview)
* Content Source: [windows-driver-docs-pr/devtest/computerhardwareids-overview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/computerhardwareids-overview.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-03-01:
@robunoki-microsoft - Thanks for this issue, I updated and reorganized the three topics into one: 

https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/computerhardwareids

The new topic shows the current location in the WDK in the bin directory. 

As you work with this tool, let me know if you see anything else that can be improved. And thanks for your help in improving the docs.

closed 2023-03-01 by DOMARS

robunoki-microsoft, 2023-03-01:
Much improved - thank you :)