MicrosoftDocs/windows-driver-docs #3418: Clarification needed on whether paging path constraints also applies to IRPs sent by IoCallDriver

wmjdgla, 2023-02-22:
> ## IRQL-Related Constraints
> Note: For information about which types of IRPs are used in paging I/O, see [Dispatch Routine IRQL and Thread Context](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/dispatch-routine-irql-and-thread-context.md).
> - Dispatch routines in the paging I/O path should never call [IoCallDriver](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-hardware/drivers/ddi/wdm/nf-wdm-iocalldriver) at any IRQL above APC_LEVEL. If a dispatch routine raises IRQL, it must lower it before calling IoCallDriver.
> - Dispatch routines in the paging path, such as read and write, cannot safely call any kernel-mode routines that require callers to be running at IRQL PASSIVE_LEVEL.

Suppose we're in the paging read path. Per 2nd point above, we cannot call [ZwCreateFile](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-zwcreatefile) as it requires PASSIVE_LEVEL. However, the 1st point above seems to imply that we can instead roll our own CreateFile IRP (IRP_MJ_CREATE) using IoCallDriver. But, the [Dispatch Routine IRQL and Thread Context](https://learn.microsoft.com/en-us/windows-hardware/drivers/ifs/dispatch-routine-irql-and-thread-context) page states that the maximum allowed IRQL for Create dispatch routines is PASSIVE_LEVEL, which seems to violate the 2nd point.

So is sending our own IRPs via IoCallDriver a valid workaround for the 2nd point? Or are the IRPs we can send also constrained by the 2nd point and the maximum allowed IRQL of their dispatch routines?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 854abc51-a990-3990-997e-2755acb4c7f7
* Version Independent ID: 7db8ed7c-65fd-dabf-2611-82704fbd5fd3
* Content: [Constraints on Dispatch Routines - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ifs/constraints-on-dispatch-routines)
* Content Source: [windows-driver-docs-pr/ifs/constraints-on-dispatch-routines.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/constraints-on-dispatch-routines.md)
* Product: **windows-hardware**
* Technology: **ifs**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

neochristi, 2023-02-24:
To be clear, you can not issue other IRPs from a paging IO path.  You can queue a worker thread that issues IO but you **must** not wait synchronously for that worker thread to complete.  If you do you will cause deadlocks.  The answer to your last question is no, I have just provided the only way to do this.

The first bullet simply says you can't do paging IO (or any IO) at dispatch level because you will never see the operation completed.  This is because process IO completion is processed via APCs.

I think the 2nd bullet is self-explanatory, you can't lower IRL.

craigbarkhouse, 2023-02-27:
The constraint that you should never call IoCallDriver above APC_LEVEL does not automatically imply that it is safe to call IoCallDriver at APC_LEVEL in all situations.  Whether you may be at APC_LEVEL or must be at PASSIVE_LEVEL is detailed in [Dispatch Routine IRQL and Thread Context](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/dispatch-routine-irql-and-thread-context.md).  Since it is based on the type of dispatch routine that ends up getting called (e.g. Create), the specific constraints are the same whether you call IoCallDriver directly (e.g. with an IRP_MJ_CREATE Irp) versus calling a higher level kernel routine (e.g. ZwCreateFile) that in turn calls IoCallDriver.

lorihollasch, 2023-03-16:
@wmjdgla, I've added the clarifying information that Neal and Craig provided to the page - hope that helps. Thanks for the issue!

closed 2023-03-16 by lorihollasch