MicrosoftDocs/windows-driver-docs #3424: dll and sys files Which files should be created?

hongyver, 2023-03-02:
It says .sys file is created, but .dll file is created.
So I create a new project from a template like Empty WDM Driver. And just added the .def file.
Then the sys file is created and it works normally.

Which method is correct? Are both the correct ways?
Is the sys file correct? Is the dll file correct?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 6d33e947-6ef6-456f-784c-6635929151f9
* Version Independent ID: c784fe86-b359-9805-30a3-37ae986b1bef
* Content: [Creating Export Drivers - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/creating-export-drivers)
* Content Source: [windows-driver-docs-pr/kernel/creating-export-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/creating-export-drivers.md)
* Product: **windows-hardware**
* Technology: **kernel**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-03-02:
Thanks for the question @hongyver.  

We had an issue about the same documentation page, #1604.  In that issue, @martindrab discusses his use of export drivers.  Martin, do you have any insight on this question?

hongyver, 2023-03-07:
Hi @tedhudek.

after step 3,

1>Building 'exportdrv' with toolset 'WindowsKernelModeDriver10.0' and the 'Desktop' target platform.
1>main.cpp
1>exportdrv.vcxproj -> D:\project\exportdrv\x64\Debug\exportdrv.dll
1>Inf2Cat task was skipped as there were no inf files to process

after step 4,

1>Building 'exportdrv' with toolset 'WindowsKernelModeDriver10.0' and the 'Desktop' target platform.
1>main.cpp
1>   D:\project\exportdrv\x64\Debug\exportdrv.lib 라이브러리 및 D:\project\exportdrv\x64\Debug\exportdrv.exp 개체를 생성하고 있습니다.
1>exportdrv.vcxproj -> D:\project\exportdrv\x64\Debug\exportdrv.dll
1>Inf2Cat task was skipped as there were no inf files to process

There is no difference than the ExportDrv.lib file is created.

tedhudek, 2023-03-07:
Hi @hongyver,

Ok so if you do steps 1-3, you get a dll file and no lib file.
And if you do steps 1-4, you get a dll file and a lib file.

What steps give you the sys file? Just steps 1 and 4 (but not 2 and 3)?

While technically both dll and sys are correct, for consistency, we recommend that export drivers be .sys files.

hongyver, 2023-03-08:
hi @tedhudek 

MSDN :
https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/creating-export-drivers#building-an-export-driver

If I make an export driver as written in MSDN, a dll file is created, not a sys file.
However, MSDN says that a sys file is created.

Is this what MSDN says incorrectly?

If you want to create a sys file, follow my method above to create a sys file.

Of course both work fine.

tedhudek, 2023-03-11:
Hi @hongyver, yes, you have discovered a bug on the page.

I've opened #3427 to update the page.  The goal of https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/creating-export-drivers#building-an-export-driver is to create a .sys file.

Can you please confirm that the new steps in the PR produce a .sys file for you?  If not, feel free to push a commit to the source branch, or tell me here what to change.

Thanks!

hongyver, 2023-03-13:
I don't know this process, but I added a single comment to the code on github first.

hongyver, 2023-03-13:
I don't know if it's correct to do this.
I forked the repository and made a full request after editing it.

hongyver, 2023-03-14:
Hi @tedhudek, It's thanks to you
This was my first pull request in my life.

The documentation looks good to me too.

tedhudek, 2023-03-15:
Thanks again @hongyver, come back and see us again sometime.

closed 2023-03-15 by tedhudek