MicrosoftDocs/windows-driver-docs #3426: Is PDB File Required for Attestation Signing?  CAT?

dwkaiser, 2023-03-10:
In the "Create the CAB File Section" of this page it is stated, "A typical CAB file submission must contain the following:  ...  The symbol file that is used for debugging information ...".

Each of the examples that follow on the page do not contain PDB files in the CAB submission, only the INF and SYS files.  I would expect that the CAB submission would require the CAT file, not included in the examples.  However, I would not expect the PDB file to be included in the CAB submission unless it was normally included as part of the INF package.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 674c6417-5a69-8fd5-be62-e4e06ba465be
* Version Independent ID: 674c6417-5a69-8fd5-be62-e4e06ba465be
* Content: [Attestation sign Windows 10+ drivers - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-attestation)
* Content Source: [windows-driver-docs-pr/dashboard/code-signing-attestation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/code-signing-attestation.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

dwkaiser, 2023-03-10:
This issue with the PDB was raised a few weeks ago.  To remove the ambiguity, I think that the examples need to be updated, and an explanation for the requirements added.

anaharris-ms, 2023-03-13:
Hi @dwkaiser, thanks for contacting Microsoft.

I will update the document with the explanation as given here:
https://github.com/MicrosoftDocs/windows-driver-docs/issues/3372

anaharris-ms, 2023-03-13:
Updated in https://github.com/MicrosoftDocs/windows-driver-docs-pr/pull/1893

dwkaiser, 2023-03-14:
@anaharris-ms, I am not able to view the specific PR link you referenced above; therefore, I cannot "review it".

I do think that requiring symbol files will create concerns for OEM's and their need to protect proprietary intellectual property.  I hope that the documentation will address this concern.

Thank you.